ScanOC/trunk-player #20: Moderated Register

kcwebby, 2017-03-14:
It would be nice for users to have to be moderated (or approved) after registration, or at least have a capability to enable this.   This would enable for Staff to approve/deny persons access to the site, prior to them giving access.

dreinhold, 2017-03-14:
Sounds good. I had started down the path of user models where we can give different users different access. This plays right into that it, would just be how much if any access you give a new user. The you bump them to the next access level if you want to.

Other things like some talk-groups should not by default be visible to everyone.
On mine I have the base hospital TGs set to non public (i don't see a need for people to get that).
Everything SO/PD on my system is encrypted so I don't have to worry about SWAT or anything like that.

User access is one of the things I need to tackle on my system so will probably be the next thing I work on.

dreinhold, 2017-09-04:
Does the way the new access groups work (or how they look to work) resolve this?
You can setup a default group with little or no access then people can be given additional access..

I was thinking of adding the option to get an email when a new user is created.

dreinhold, 2017-09-11:
Calling this one done with the email and new access groups.

closed 2017-09-11 by dreinhold