ScanOC/trunk-player #21: Anonymous Listening - Disable

kcwebby, 2017-03-14:
I think it would be nice to be able to set the Anonymous_Time setting to something that completely disables it.

Currently, I have mine set to 0.00001, which essentially does that, but doesn't quite fit the bill as the message states "You are not currently logged into the system. You will only see the last 1e-05 minutes of history. Login or Create a free acount to see more history."

dreinhold, 2017-03-14:
Use `ANONYMOUS_TIME = -1` to disable anonymous access.
The message then shows :
"You are not currently logged into the system. Login to access the system."

closed 2017-03-14 by dreinhold

imagesafari, 2019-02-08:
This doesn't seem to work in the latest release. What is the easiest way to require a login to see anything?

dreinhold, 2019-02-08:
Try adding to your local settings
OPEN_SITE = False

imagesafari, 2019-02-08:
That does not seem to make a difference