ScanOC/trunk-player #23: Call Menu - irregularities

kcwebby, 2017-03-14:
I've noticed when screening calls, its not possible to get the call menu to display while new transmissions are loading in (when queuing up)... Perhaps it would be better to remove the menu and have two buttons for Hold on Talkgroup, and Details

I also think clicking on the Talkgroup could be modified to lock in on that talkgroup as well, or show the details (no preference).

dreinhold, 2017-03-14:
Yeah I think that menu's job is to train you to click fast ;)

That was something I wanted to tackle last week but ended up adding the multi system support ;(

In the old UI there were more options that I have not had time to re implement so the menu had more in it before and it became crowed when there were all just buttons on the screen.

My quick fix I wanted to do was just keep the re fresh from happening while the menu was open.

RachelOlivero, 2017-04-07:
@dreinhold I'm happy to help with this one. I see two solutions.

Short term: implement pause loading while menu is open. Implement timeout to close the menu after x seconds if the user popped it open and then didn't click.

Long term: When I took a quick look at the code, it looks like the table is being rebuilt each time the transmissions refresh. If that's the case, maybe changing it so when a new transmission comes in it is prepended to the table and the last one on the page removed would allow the menu to stay visible, even though it would "slide" down. That's just based on a quick look at the code.

I don't want to step on your toes if you're already working on this, but I thought I'd offer in case you've got other priorities for your time.

dreinhold, 2017-04-07:
I would agree on all points.
Go for it.

closed 2017-05-15 by dreinhold