ScanOC/trunk-player #24: Enh: "Loading Data"

kcwebby, 2017-04-16:
"Loading Data..." will display on a scan list even if the user is not logged in, and the ANONYMOUS_TIME = 0;   This might lead people to believe that the website is not functioning.  I think it would be good to remove this if we are not in fact open to socket for data.

dreinhold, 2017-04-16:
I probably need to change it when ANONYMOUS_TIME = 0 to just kick the person back to the sign-on page.

dreinhold, 2017-04-23:
@kcwebby maybe I misunderstood this.  If ANONYMOUS_TIME = 0 and he user is not logged in, it means they can see everything. So if it's stuck on Loading data, that does mean the site/page is having issues.

I was thinking ANONYMOUS_TIME = -1 when I replied. I was going to add code when its -1 to redirect back to the login page, but not sure this is what you are asking about.

kcwebby, 2017-04-24:
Doh...

Yes.. I meant -1, I don't know why I typed 0..

Sorry about that, you are on the right page.

I was intending for someone not logged in to see loading data... if the
anon_time was disabled with -1.

dreinhold, 2017-04-26:
I changed the JS, when the json data returns if it is empty it will clear the loading data line.

closed 2017-05-15 by dreinhold