ScanOC/trunk-player #28: Transmissions Page does not display System

kcwebby, 2017-05-18:
/audio/5eac024b-ea2a-4f70-a3de-ca3b478ac933/ (or equiv)  URL details does not display source system.

On a side note, it does display something called "Source"  But I'm not sure what that field /table is to be used for.

dreinhold, 2017-05-18:
Oh good catch, yeah easy one to fix.

As for source, this is a spot to track which sdr stick captured the traffic.
if you inject source: and a number in the json it will record, or pass --source {number} in the add_transmission command line.
I modified trunk record to add it, I was trying to track down where some static was coming from, so I could look and see which recorder it came from when I heard the static.

kcwebby, 2017-05-19:
OO, I would love to do that to my copy of trunk recorder..  can you point
me to what to change to make that work? (keep in mind I have treehouse's
ncurses version.

dreinhold, 2017-09-04:
The system was added in commit 325bade664fd8cd7ea4f362f5e63bb074398d0d6

closed 2017-09-04 by dreinhold