ScanOC/trunk-player #31: Fresh install fails during ./manage.py migrate

dancorr, 2017-09-29:
Applying radio.0035_add_anonymous_user...Traceback (most recent call last):
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/sqlite3/base.py", line 328, in execute
    return Database.Cursor.execute(self, query, params)
sqlite3.OperationalError: no such column: radio_talkgroupaccess.default_new_talkgroups

dreinhold, 2017-09-29:
Oh, I got that same error the other day when I was working on some updates, I assumed it was due to the way I was rolling back test data. Guess not...
Taking a look

dreinhold, 2017-09-29:
I just pushed fce3d0e4 which should fix it. Just got another git pull and retry the migration.

dancorr, 2017-09-29:
Fixed! Can't wait to get this up and running. Thanks!

dreinhold, 2017-09-29:
So fair warning the documentation is pretty poor. So please feel free to speak up if something just does not seem right.

closed 2017-09-29 by dreinhold