ScanOC/trunk-player #32: Apache Reverse Proxy WebSocket issues

MaxwellDPS, 2017-10-03:
I have my setup as follows:
 Trunkplayer(172.16.3.40) -NO SSL-> Apache Server as Reverse Proxy -SSL-> Radio.maxwelldps.com

The HTML Works fine... But over https://radio.maxwelldps.com/ the web sockets give the error:

> WebSocket connection to 'wss://radio.maxwelldps.com/ws-calls/scan/default/' failed: WebSocket is closed before the connection is established.

However Locally It works fine...

This may be my apache config... but I'm not sure.
Apache config:
```
<VirtualHost _default_:443>
		ServerAdmin none@none.io
		ServerName radio.maxwelldps.com
		ErrorLog ${APACHE_LOG_DIR}/error.log
		CustomLog ${APACHE_LOG_DIR}/access.log combined
		SSLEngine on
		SSLCertificateFile	/etc/apache2/ssl/MaxwellDPS.crt
		SSLCertificateKeyFile /etc/apache2/ssl/ssl.key
		SSLCertificateChainFile /etc/apache2/ssl/int.crt	
		
		ProxyPassMatch /ws-calls/(.*)$ ws://172.16.3.40/$1		
		ProxyPassReverse /ws-calls/ ws://172.16.3.40/ws-calls/
		
		
		ProxyPass / http://172.16.3.40/
		ProxyPassReverse / http://172.16.3.40/

	</VirtualHost>
```

Reverse Proxy:
[image]

Local Direct Access:
[image]

dreinhold, 2017-10-03:
I don't use apache, but looking at some docs for using apac he with django channels you could try.

Replace
```
    ProxyPassMatch /ws-calls/(.*)$ ws://172.16.3.40/$1		
    ProxyPassReverse /ws-calls/ ws://172.16.3.40/ws-calls/

```
With
```
  ProxyPass "/ws-calls/" "ws://172.16.3.40/"
  ProxyPassReverse "/ws-calls/" "ws://172.16.3.40/ws-calls/"
```

MaxwellDPS, 2017-10-03:
Hmmm... I still am getting the error 

> WebSocket connection to 'wss://radio.maxwelldps.com/ws-calls/scan/default/' failed: WebSocket is closed before the connection is established.

I was using this line to match all the different tg and scan urls
`ProxyPassMatch /ws-calls/(.*)$ ws://172.16.3.40/$1`

MaxwellDPS, 2017-10-03:
feel free to visit radio.maxwelldps.com and check the debug channel

Im still seeing:
[image]

dreinhold, 2017-10-03:
Do you have the daphne running to service the web sockets? Also normally it's not serving then on port 80. Mine is on port 7055. So check the port daphne is runnging and point proxy pass to that ip:port

MaxwellDPS, 2017-10-03:
Yep I have daphne Running... but it looks like the ws call are still over port 80 Locally
I tried this:
```
	ProxyPassMatch /ws-calls/(.*)$ ws://172.16.3.40:7055/$1		
		ProxyPassReverse /ws-calls/ ws://172.16.3.40:7055/ws-calls/ 
```

That gave me:
[image]

MaxwellDPS, 2017-10-03:
Here is a local request ... it appears to be on port 80
[image]

dreinhold, 2017-10-03:
When you start daphine what does the startup line show : 
`Starting server at tcp:port=7055:interface=127.0.0.1, channel layer trunk_player.asgi:channel_layer.`

*I am leaving for a few hours but will check when I get back.

MaxwellDPS, 2017-10-03:
Sure thing Thanks:
heres it output Lake is local... default is over proxy
```
2017-10-03 11:57:58,755 INFO     Starting server at tcp:port=7055:interface=127.0.0.1, channel layer trunk_player.asgi:channel_layer.
2017-10-03 11:57:58,757 INFO     HTTP/2 support not enabled (install the http2 and tls Twisted extras)
2017-10-03 11:57:58,757 INFO     Using busy-loop synchronous mode on channel layer
2017-10-03 11:57:58,757 INFO     Listening on endpoint tcp:port=7055:interface=127.0.0.1
127.0.0.1:41336 - - [03/Oct/2017:11:57:59] "WSCONNECTING /ws-calls/tg/lake-county-so-1/" - -
127.0.0.1:41336 - - [03/Oct/2017:11:58:00] "WSCONNECT /ws-calls/tg/lake-county-so-1/" - -
127.0.0.1:41338 - - [03/Oct/2017:11:58:00] "WSCONNECTING /ws-calls/scan/default/" - -
127.0.0.1:41338 - - [03/Oct/2017:11:58:01] "WSCONNECT /ws-calls/scan/default/" - -
127.0.0.1:41340 - - [03/Oct/2017:11:58:18] "GET /api_v1/tg/lake-county-so-1/" 200 50965
127.0.0.1:41342 - - [03/Oct/2017:11:58:19] "GET /api_v1/scan/default/" 200 42052
127.0.0.1:41346 - - [03/Oct/2017:11:58:26] "GET /scan/default/" 200 5774
127.0.0.1:41344 - - [03/Oct/2017:11:58:26] "GET /api_v1/scan/default/" 200 42052
127.0.0.1:41350 - - [03/Oct/2017:11:58:27] "GET /api_v1/menutalkgrouplist/" 200 214
127.0.0.1:41352 - - [03/Oct/2017:11:58:27] "GET /api_v1/menuscanlist/" 200 261
127.0.0.1:41348 - - [03/Oct/2017:11:58:27] "GET /api_v1/scan/default/" 200 42052

```

MaxwellDPS, 2017-10-03:
Ok, I GOT IT TO WORK!

It was just an Apache issue :)
[image]

For anyone who wants to reverse proxy this is what worked for me
```
<VirtualHost _default_:443>
		ServerAdmin none@none.io
		ServerName radio.maxwelldps.com
		ErrorLog ${APACHE_LOG_DIR}/error.log
		CustomLog ${APACHE_LOG_DIR}/access.log combined
		SSLEngine on
		SSLCertificateFile	/etc/apache2/ssl/MaxwellDPS.crt
		SSLCertificateKeyFile /etc/apache2/ssl/ssl.key
		SSLCertificateChainFile /etc/apache2/ssl/int.crt	
		
		ProxyPassMatch ^/ws-calls/(.*)$ ws://172.16.3.40/ws-calls/$1		
		ProxyPassReverse ^/ws-calls/ ws://172.16.3.40/ws-calls/
		
		
		ProxyPass / http://172.16.3.40/
		ProxyPassReverse / http://172.16.3.40/

	</VirtualHost>
```

closed 2017-10-03 by MaxwellDPS

dreinhold, 2017-10-03:
Very cool. Thanks for posting your working config.