ScanOC/trunk-player #33: Feature Request: Sort by Date, length, tg, etc...

MaxwellDPS, 2017-11-01:
I think it would be great to add sorting on scanlists!
by some of theses... Let me know if you have more below

- Date/time

- length

- Talk group

- System

dreinhold, 2017-11-16:
Not sure on this one.

So if I sort by length would it sort ALL transmissions by length?
Talkgroup wouldn't that just be the talkgroup page?

System yeah I could see that. But really that could just be a scan list.

MaxwellDPS, 2017-11-16:
Yeah, my bad in retrospect... I would like to see a way to filter in a scan list or talk group by date/and time... instead of having to go back 12 or 20 pages :) also a way to filter by unit ID would be great too.

dreinhold, 2017-11-17:
There is an undocumented feature to filter by units/

/unit/<name-of-radio-id>/

For the name you need the "slug" from the unit setting (just the name in lowercase and [ - ] for any spaces.

dreinhold, 2017-11-19:
So this is more of an advanced search feature, which yes we need...

closed 2024-03-21 by dreinhold