ScanOC/trunk-player #34: Feature Request: AAC HE m4a instead of LAME mp3

MaxwellDPS, 2017-11-01:
What would have to be modified to allow for AAC HE 1/2 m4a files instead of mp3?

Just given that a ~100mb mp3 is ~60mb /w AAC HEv1 and ~20mb /w AAC HEv2
Also m4a Has great support in all major browsers, Linux, OSX, and iOS, Android Support <- Not to mention less data transfer

In terms of scripts would it be possible to modify manage.py to look for a m4a instead of mp3 when it reads the json? As I think this would be the easiest

dreinhold, 2017-11-01:
When I first started I was using m4a's and something was tripping me up. So I moved to mp3.
I don't remember what was the issue. But yeah it's just a change in a few spots to enable m4a support.

MaxwellDPS, 2017-11-01:
GREAT I Will play around with the encoding command for the script

MaxwellDPS, 2017-11-01:
I got FFMPEG to take a t2.wav which is a 4 sec real audio file to this... Mp3 For scale
[image]

AAC HE V1 Only - Mono audio
FFMPEG command I have had luck with... I'm sure it could be fine tunned

`ffmpeg -i $filename -c:a libfdk_aac -profile:a aac_he -b:a 32k -ar 48k $AACencoded`

```
#! /bin/bash
# Script to upload audio from trunk-recorder to 
# web server to be displaied by trunk-player
#
# 02/15/2016 - Dylan Reinhold dreinhold@gmail.com
#-----------------------------------------------------
#
REMOTE_USER_NAME="radio"
REMOTE_SERVER="0.0.0.0"
REMOTE_AUDIO_FOLDER="/home/$REMOTE_USER_NAME/audio"
REMOTE_IMPORT_SCRIPT="/home/$REMOTE_USER_NAME=/trunk-player/utility/server/encode_load.sh"
echo "Encoding: $1" 
filename="$1"
basename="${filename%.*}"
filename_only=$(basename $basename)
json="$basename.json"
AACencoded="$basename.m4a"
web_dir=$(dirname $filename | cut -d/ -f4-)"/"
system=0 # Change this for each system

# Hack the JSON to add play length and source
len=$(soxi -D $filename)

head -n-2 $json > $json.new
echo "\"play_length\": $len," >> $json.new
echo "\"source\": 0," >> $json.new
tail -n2 $json >> $json.new
mv $json.new $json

ffmpeg -i $filename -c:a libfdk_aac -profile:a aac_he -b:a 32k -ar 48k $AACencoded

ssh -q $REMOTE_USER_NAME@$REMOTE_SERVER "cd /home/radio/trunk-player; . env/bin/activate; ./manage.py add_transmission $basename --web_url=$web_dir --system=$system"
echo "[+] UPLOADED AND ADDED"
rm -f $filename 
/home/max/st.py $AACencoded
```

dreinhold, 2017-11-11:
I just pushed a m4a_test branch, checkout that branch.
When you run add_transmission include `--m4a` to set the file as a m4a.

MaxwellDPS, 2017-11-12:
I'll Migrate over to it When I can

dreinhold, 2017-11-12:
Sounds good. I just pushed a few tweaks to speed up the migrate when creating the new database fields.
**If anyone pulled and applied this branch before now, before you pull these updates you should run `./manage.py migrate radio 0045 ` then pull the new code and run a fresh `./manage.py migrate` also don't forget to run a `./manage.py collectstatic --noinput` to update the javascript files.

MaxwellDPS, 2017-11-17:
it seems to not put the .m4a on... I am running main... ands the --m4a

MaxwellDPS, 2017-11-17:
[image]

dreinhold, 2017-11-17:
If you look at that transmission in the admin area, is the Audio file type
filled in?

MaxwellDPS, 2017-11-17:
also just noticed I keep getting logged out?
[image]

MaxwellDPS, 2017-11-17:
I cant see them in the admin area, but I can see the file type field.
I do keep getting logged out, and the transmissions will disappear on and off the scanlist

dreinhold, 2017-11-17:
For the m4a On your site run
./manage.py collectstaticfiles

It needs to refresh the javascript files.

Are you running a test-server on your same server? They might be trying to
use the same redis channels.

dreinhold, 2017-11-19:
Did you get this one working?

dreinhold, 2017-11-20:
Should be usable with commit 8d9cc42b82

closed 2017-11-20 by dreinhold