ScanOC/trunk-player #35: Unable to start the server

cmnybo, 2017-11-04:
I installed trunk-player according to the directions on the installation page. Everything worked correctly up until the `./manage.py runserver` command.
I get a bunch of connection refused errors and the server at localhost:8000 is not accessible.
```
Performing system checks...

System check identified no issues (0 silenced).
November 04, 2017 - 15:34:30
Django version 1.11.1, using settings 'trunk_player.settings'
Starting Channels development server at http://127.0.0.1:8000/
Channel layer default (asgi_redis.core.RedisChannelLayer)
Quit the server with CONTROL-C.
2017-11-04 15:34:30,553 - INFO - worker - Listening on channels http.request, websocket.connect, websocket.disconnect, websocket.receive
2017-11-04 15:34:30,554 - INFO - worker - Listening on channels http.request, websocket.connect, websocket.disconnect, websocket.receive
2017-11-04 15:34:30,555 - INFO - worker - Listening on channels http.request, websocket.connect, websocket.disconnect, websocket.receive
2017-11-04 15:34:30,556 - INFO - worker - Listening on channels http.request, websocket.connect, websocket.disconnect, websocket.receive
2017-11-04 15:34:30,557 - INFO - server - HTTP/2 support not enabled (install the http2 and tls Twisted extras)
2017-11-04 15:34:30,558 - INFO - server - Using busy-loop synchronous mode on channel layer
2017-11-04 15:34:30,558 - INFO - server - Listening on endpoint tcp:port=8000:interface=127.0.0.1
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 439, in connect
    sock = self._connect()
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 494, in _connect
    raise err
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 482, in _connect
    sock.connect(socket_address)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/client.py", line 572, in execute_command
    connection.send_command(*args)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 563, in send_command
    self.send_packed_command(self.pack_command(*args))
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 538, in send_packed_command
    self.connect()
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 442, in connect
    raise ConnectionError(self._error_message(e))
redis.exceptions.ConnectionError: Error 111 connecting to localhost:6379. Connection refused.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 439, in connect
    sock = self._connect()
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 494, in _connect
    raise err
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 482, in _connect
    sock.connect(socket_address)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/channels/management/commands/runserver.py", line 166, in run
    worker.run()
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/channels/worker.py", line 87, in run
    channel, content = self.channel_layer.receive_many(channels, block=True)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/asgiref/base_layer.py", line 43, in receive_many
    return self.receive(channels, block)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/asgi_redis/core.py", line 211, in receive
    result = connection.blpop(list_names, timeout=self.blpop_timeout)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/client.py", line 1163, in blpop
    return self.execute_command('BLPOP', *keys)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/client.py", line 578, in execute_command
    connection.send_command(*args)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 563, in send_command
[... 161 lines cut ...]
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/channels/management/commands/runserver.py", line 166, in run
    worker.run()
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/channels/worker.py", line 87, in run
    channel, content = self.channel_layer.receive_many(channels, block=True)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/asgiref/base_layer.py", line 43, in receive_many
    return self.receive(channels, block)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/asgi_redis/core.py", line 211, in receive
    result = connection.blpop(list_names, timeout=self.blpop_timeout)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/client.py", line 1163, in blpop
    return self.execute_command('BLPOP', *keys)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/client.py", line 578, in execute_command
    connection.send_command(*args)
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 563, in send_command
    self.send_packed_command(self.pack_command(*args))
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 538, in send_packed_command
    self.connect()
  File "/home/cody/trunk-player/env/lib/python3.5/site-packages/redis/connection.py", line 442, in connect
    raise ConnectionError(self._error_message(e))
redis.exceptions.ConnectionError: Error 111 connecting to localhost:6379. Connection refused.

2017-11-04 15:34:35,567 - ERROR - server - Error trying to receive messages: Error 111 connecting to localhost:6379. Connection refused.`
```

dreinhold, 2017-11-04:
That is the site unable to connect to the redis server. Make sure redis is
up and listening on port 6379.

Dylan

cmnybo, 2017-11-04:
OK, it's working now. I didn't have redis started.

closed 2017-11-04 by cmnybo