ScanOC/trunk-player #37: Database is locked causing transmission drop and 500 Error

MaxwellDPS, 2017-11-15:
I keep getting this error when I have multiple people on the site or when many transmissions hit at once... 

```
Traceback (most recent call last):
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/sqlite3/base.py", line 328, in execute
    return Database.Cursor.execute(self, query, params)
sqlite3.OperationalError: database is locked
```

I read this [https://stackoverflow.com/questions/3172929/operationalerror-database-is-locked](https://stackoverflow.com/questions/3172929/operationalerror-database-is-locked)
and I think that we should consider PostgreSQL instead of sqlite... 
It seems the sqlite is very limited on concurent use... since radio transmissions come in very quick I think the migration to PostgreSQL would solve this issue...

django seems to have this built in too:
https://docs.djangoproject.com/en/dev/ref/databases/#database-is-locked-errorsoption

```
Traceback (most recent call last):
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/sqlite3/base.py", line 328, in execute
    return Database.Cursor.execute(self, query, params)
sqlite3.OperationalError: database is locked

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 363, in execute_from_command_line
    utility.execute()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 355, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 283, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 330, in execute
    output = self.handle(*args, **options)
  File "/home/radio/trunk-player/radio/management/commands/add_transmission.py", line 53, in handle
    add_new_trans(options)
  File "/home/radio/trunk-player/radio/management/commands/add_transmission.py", line 106, in add_new_trans
    source = Source.objects.get(pk=source_opt)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/query.py", line 374, in get
    num = len(clone)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/query.py", line 232, in __len__
    self._fetch_all()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/query.py", line 1103, in _fetch_all
    self._result_cache = list(self._iterable_class(self))
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 886, in execute_sql
    raise original_exception
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 876, in execute_sql
    cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/utils.py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/sqlite3/base.py", line 328, in execute
    return Database.Cursor.execute(self, query, params)
django.db.utils.OperationalError: database is locked

```

dreinhold, 2017-11-15:
> errorsoption

Yes. I was going to point you to the instructions to use postgress, which i
don't see. Hmm failure on my part.

I also wrote some migration code to move you data from sqlite into postgres.

There was coversation on the google mailing list, but i never added it to
the project documentation.

Ill get the info tonight.

MaxwellDPS, 2017-11-15:
Great! Thanks

dreinhold, 2017-11-16:
This was the message I sent to the google groups list.
https://groups.google.com/forum/#!topic/trunk-player/8SPJCRNYXRo

Now the database has changed from that point and my migrate script probably needs to be updated.
The good thing about he sqlite is it is just that one file you need to backup `db.sqlite3`. Make a copy of that file if you try the conversion (really should not hurt the current DB) just your new postgres one if it goes wrong.

In the email it has the config settings you need to make to connect to postgres vs sqlite.
I don't know if I will have time to try it tonight, but ill try a test migration soon and update the script if needed.

MaxwellDPS, 2017-11-16:
I just tried the migrate command and hers the output

```
radio@FOX3:~$ cd t*
radio@FOX3:~/trunk-player$ ls
audio_files         docs         local_override  requirements.txt  trunk_player
daphne.sh           env          manage.py       run.sh            utility
db.sqlite3          icons        radio           server.sh         worker.sh
db.sqlite3-journal  License.txt  README.md       static
radio@FOX3:~/trunk-player$ cp db.sqlite3 ../.
radio@FOX3:~/trunk-player$ ls ../
db.sqlite3  trunk-player
radio@FOX3:~/trunk-player$ nano t*/*ocal.py
radio@FOX3:~/trunk-player$ source env/bin/activate
(Trunk Player)radio@FOX3:~/trunk-player$ ./manage.py migrate
Operations to perform:
  Apply all migrations: account, admin, auth, contenttypes, pinax_stripe, radio,                                                                                         sessions, sites, socialaccount
Running migrations:
  Applying contenttypes.0001_initial... OK
  Applying auth.0001_initial... OK
  Applying account.0001_initial... OK
  Applying account.0002_email_max_length... OK
  Applying admin.0001_initial... OK
  Applying admin.0002_logentry_remove_auto_add... OK
  Applying contenttypes.0002_remove_content_type_name... OK
  Applying auth.0002_alter_permission_name_max_length... OK
  Applying auth.0003_alter_user_email_max_length... OK
  Applying auth.0004_alter_user_username_opts... OK
  Applying auth.0005_alter_user_last_login_null... OK
  Applying auth.0006_require_contenttypes_0002... OK
  Applying auth.0007_alter_validators_add_error_messages... OK
  Applying auth.0008_alter_user_username_max_length... OK
  Applying pinax_stripe.0001_initial... OK
  Applying pinax_stripe.0002_auto_20151205_1451... OK
  Applying pinax_stripe.0003_make_cvc_check_blankable... OK
  Applying pinax_stripe.0004_plan_metadata... OK
  Applying pinax_stripe.0005_auto_20161006_1445... OK
  Applying pinax_stripe.0006_coupon... OK
  Applying pinax_stripe.0007_auto_20170108_1202... OK
  Applying radio.0001_initial... OK
  Applying radio.0002_unit... OK
  Applying radio.0003_auto_20160605_1043... OK
  Applying radio.0004_auto_20160614_2235... OK
  Applying radio.0005_talkgroup_slug... OK
  Applying radio.0006_talkgroup_common_name... OK
  Applying radio.0007_scanlist... OK
  Applying radio.0008_transmission_play_length... OK
  Applying radio.0009_auto_20160830_1951... OK
  Applying radio.0010_transmission_trans_id... OK
  Applying radio.0011_auto_20160909_1532... OK
  Applying radio.0012_auto_20161008_2103... OK
  Applying radio.0013_auto_20161012_1515... OK
  Applying radio.0014_auto_20161012_1530... OK
  Applying radio.0015_auto_20161014_2051... OK
  Applying radio.0016_default_agency... OK
  Applying radio.0017_auto_20161014_2225... OK
  Applying radio.0018_auto_20161023_1622... OK
  Applying radio.0019_talkgroup_comments... OK
  Applying radio.0020_plan_profile... OK
  Applying radio.0021_default_plan... OK
  Applying radio.0022_system... OK
  Applying radio.0023_default_system... OK
  Applying radio.0024_auto_20170225_0857... OK
  Applying radio.0025_unit_slug... OK
  Applying radio.0026_auto_20170305_1336... OK
  Applying radio.0027_auto_20170306_2118... OK
  Applying radio.0028_auto_20170307_2152... OK
  Applying radio.0029_scanlist_slug... OK
  Applying radio.0030_scanlist_build_slug... OK
  Applying radio.0031_webhtml... OK
[... 51 lines cut ...]
  File "/home/radio/trunk-player/radio/models.py", line 394, in create_profile
    for tg in TalkGroupAccess.objects.filter(default_group=True):
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/model                                                                                        s/query.py", line 250, in __iter__
    self._fetch_all()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/model                                                                                        s/query.py", line 1103, in _fetch_all
    self._result_cache = list(self._iterable_class(self))
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/model                                                                                        s/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/model                                                                                        s/sql/compiler.py", line 886, in execute_sql
    raise original_exception
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/model                                                                                        s/sql/compiler.py", line 876, in execute_sql
    cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backe                                                                                        nds/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/utils                                                                                        .py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/utils/si                                                                                        x.py", line 685, in reraise
    raise value.with_traceback(tb)
[... cut ...] ke                                                                                        nds/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
django.db.utils.ProgrammingError: column radio_talkgroupaccess.default_new_talkg                                                                                        roups does not exist
LINE 1: ...."name", "radio_talkgroupaccess"."default_group", "radio_tal...
                                                             ^

```

dreinhold, 2017-11-16:
I assume this is the most current version?

I feel like I have fixed this a couple times, I must be missing something
in here.
Ill take a look.

Dylan

MaxwellDPS, 2017-11-16:
Yes, I have not switched to the m4a branch yet. Thanks

MaxwellDPS, 2017-11-17:
If you want I can post the contents of the postgresql db

dreinhold, 2017-11-17:
I will probably be able to re create it. Will not be able to look until
later tonight.

dreinhold, 2017-11-17:
Okay recreated it with fresh postgres install.
Right now I can think of 3 bad ways to fix it. But need to work on a good
way ;)

dreinhold, 2017-11-17:
I just pushed 01c9047c8 this should fix this issue in the migrate...

MaxwellDPS, 2017-11-17:
Sorry to come back like this :)
THE MIGRATE WORKS!
but move_db fails with this error
Ps. I tried a blank DB...and the user has all permissions on the DB + Superuser

```
radio@FOX3:~/trunk-player$ source env/bin/activate
(Trunk Player)radio@FOX3:~/trunk-player$ ./manage.py move_db
Moving data from app auth table group into new db
Moving data from app auth table group_permissions into new db
Moving data from app auth table permission into new db
Traceback (most recent call last):
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
psycopg2.IntegrityError: duplicate key value violates unique constraint "auth_permission_content_type_id_codename_01ab375a_uniq"
DETAIL:  Key (content_type_id, codename)=(13, add_profile) already exists.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 363, in execute_from_command_line
    utility.execute()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 355, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 283, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 330, in execute
    output = self.handle(*args, **options)
  File "/home/radio/trunk-player/radio/management/commands/move_db.py", line 54, in handle
    move_all_db_data(options)
  File "/home/radio/trunk-player/radio/management/commands/move_db.py", line 87, in move_all_db_data
    rec.save(using='default')
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 806, in save
    force_update=force_update, update_fields=update_fields)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 836, in save_base
    updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 903, in _save_table
    forced_update)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 953, in _do_update
    return filtered._update(values) > 0
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/query.py", line 662, in _update
    return query.get_compiler(self.db).execute_sql(CURSOR)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1191, in execute_sql
    cursor = super(SQLUpdateCompiler, self).execute_sql(result_type)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 886, in execute_sql
    raise original_exception
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 876, in execute_sql
    cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/utils.py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
django.db.utils.IntegrityError: duplicate key value violates unique constraint "auth_permission_content_type_id_codename_01ab375a_uniq"
DETAIL:  Key (content_type_id, codename)=(13, add_profile) already exists.

```

MaxwellDPS, 2017-11-17:
And now on trunk player I am getting this... even after I copied my backup DB over

```
Traceback (most recent call last):
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/sqlite3/base.py", line 328, in execute
    return Database.Cursor.execute(self, query, params)
sqlite3.OperationalError: table radio_transmission has no column named audio_file_type

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 363, in execute_from_command_line
    utility.execute()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 355, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 283, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 330, in execute
    output = self.handle(*args, **options)
  File "/home/radio/trunk-player/radio/management/commands/add_transmission.py", line 60, in handle
    add_new_trans(options)
  File "/home/radio/trunk-player/radio/management/commands/add_transmission.py", line 150, in add_new_trans
    t.save()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 806, in save
    force_update=force_update, update_fields=update_fields)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 836, in save_base
    updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 922, in _save_table
    result = self._do_insert(cls._base_manager, using, fields, update_pk, raw)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 961, in _do_insert
    using=using, raw=raw)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/query.py", line 1061, in _insert
    return query.get_compiler(using=using).execute_sql(return_id)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1099, in execute_sql
    cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/utils.py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/sqlite3/base.py", line 328, in execute
    return Database.Cursor.execute(self, query, params)
django.db.utils.OperationalError: table radio_transmission has no column named audio_file_type
```

dreinhold, 2017-11-17:
There was some DB changes in the last couple commits.
So your current sqlite3 database does not have the new layout.

So change your settings back to point to the sqlite DB and run a migrate to
update that one to the newest layout.

Then change the settings back to both DBs and re run the move db.

MaxwellDPS, 2017-11-17:
I just tried that and got

```
(Trunk Player)radio@FOX3:~/trunk-player$ ./manage.py migrate
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 363, in execute_from_command_line
    utility.execute()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 355, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 283, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 330, in execute
    output = self.handle(*args, **options)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/commands/migrate.py", line 86, in handle
    executor.loader.check_consistent_history(connection)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/migrations/loader.py", line 298, in check_consistent_history
    connection.alias,
django.db.migrations.exceptions.InconsistentMigrationHistory: Migration radio.0035_add_anonymous_user is applied before its dependency radio.0048_auto_20171112_1144 on database 'default'.
```

dreinhold, 2017-11-17:
Arg that one is give me so much trouble in so many ways...

For now try this just delete this one.

rm radio/migrations/0035_add_anonymous_user.py

Then run another migration./

MaxwellDPS, 2017-11-17:
Thanks, I remigrated... then migrated to a bland postgresql db and that went fine but on the move I got

```
(Trunk Player)radio@FOX3:~/trunk-player$ ./manage.py move_db
Moving data from app auth table group into new db
Moving data from app auth table group_permissions into new db
Moving data from app auth table permission into new db
Traceback (most recent call last):
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
psycopg2.IntegrityError: duplicate key value violates unique constraint "auth_permission_content_type_id_codename_01ab37                                                                                                                     5a_uniq"
DETAIL:  Key (content_type_id, codename)=(37, add_customer) already exists.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 363, in execu                                                                                                                     te_from_command_line
    utility.execute()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 355, in execu                                                                                                                     te
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 283, in run_from_                                                                                                                     argv
    self.execute(*args, **cmd_options)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 330, in execute
    output = self.handle(*args, **options)
  File "/home/radio/trunk-player/radio/management/commands/move_db.py", line 54, in handle
    move_all_db_data(options)
  File "/home/radio/trunk-player/radio/management/commands/move_db.py", line 87, in move_all_db_data
    rec.save(using='default')
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 806, in save
    force_update=force_update, update_fields=update_fields)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 836, in save_base
    updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 903, in _save_table
    forced_update)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 953, in _do_update
    return filtered._update(values) > 0
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/query.py", line 662, in _update
    return query.get_compiler(self.db).execute_sql(CURSOR)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1191, in execut                                                                                                                     e_sql
    cursor = super(SQLUpdateCompiler, self).execute_sql(result_type)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 886, in execute                                                                                                                     _sql
    raise original_exception
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 876, in execute                                                                                                                     _sql
    cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/utils.py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
django.db.utils.IntegrityError: duplicate key value violates unique constraint "auth_permission_content_type_id_codename                                                                                                                     _01ab375a_uniq"
DETAIL:  Key (content_type_id, codename)=(37, add_customer) already exists.

```

dreinhold, 2017-11-17:
What version of postgres are you running?
I don't seem to get that error.

You can give this a try, edit radio/management/commands/move_db.py
Comment out these 3:
    'auth_group',
    'auth_group_permissions',
    'auth_permission',
In the table_list array and re run it. It will skip those tables.

Dylan

MaxwellDPS, 2017-11-17:
I think postgresql is the most recent version, just to double check I made a new db and got slightly different results on move_db... I dont know why this isnt working lol I build a brand new VM just to run trunk-player, postgresql... lol
```
postgres=# SELECT version();
                                                     version
------------------------------------------------------------------------------------------------------------------
 PostgreSQL 9.5.10 on x86_64-pc-linux-gnu, compiled by gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609, 64-bit
(1 row)

```

```
(Trunk Player)radio@FOX3:~/trunk-player$ ./manage.py migrate
Operations to perform:
  Apply all migrations: account, admin, auth, contenttypes, pinax_stripe, radio, sessions, sites, socialaccount
Running migrations:
  Applying contenttypes.0001_initial... OK
  Applying auth.0001_initial... OK
  Applying account.0001_initial... OK
  Applying account.0002_email_max_length... OK
  Applying admin.0001_initial... OK
  Applying admin.0002_logentry_remove_auto_add... OK
  Applying contenttypes.0002_remove_content_type_name... OK
  Applying auth.0002_alter_permission_name_max_length... OK
  Applying auth.0003_alter_user_email_max_length... OK
  Applying auth.0004_alter_user_username_opts... OK
  Applying auth.0005_alter_user_last_login_null... OK
  Applying auth.0006_require_contenttypes_0002... OK
  Applying auth.0007_alter_validators_add_error_messages... OK
  Applying auth.0008_alter_user_username_max_length... OK
  Applying pinax_stripe.0001_initial... OK
  Applying pinax_stripe.0002_auto_20151205_1451... OK
  Applying pinax_stripe.0003_make_cvc_check_blankable... OK
  Applying pinax_stripe.0004_plan_metadata... OK
  Applying pinax_stripe.0005_auto_20161006_1445... OK
  Applying pinax_stripe.0006_coupon... OK
  Applying pinax_stripe.0007_auto_20170108_1202... OK
  Applying radio.0001_initial... OK
  Applying radio.0002_unit... OK
  Applying radio.0003_auto_20160605_1043... OK
  Applying radio.0004_auto_20160614_2235... OK
  Applying radio.0005_talkgroup_slug... OK
  Applying radio.0006_talkgroup_common_name... OK
  Applying radio.0007_scanlist... OK
  Applying radio.0008_transmission_play_length... OK
  Applying radio.0009_auto_20160830_1951... OK
  Applying radio.0010_transmission_trans_id... OK
  Applying radio.0011_auto_20160909_1532... OK
  Applying radio.0012_auto_20161008_2103... OK
  Applying radio.0013_auto_20161012_1515... OK
  Applying radio.0014_auto_20161012_1530... OK
  Applying radio.0015_auto_20161014_2051... OK
  Applying radio.0016_default_agency... OK
  Applying radio.0017_auto_20161014_2225... OK
  Applying radio.0018_auto_20161023_1622... OK
  Applying radio.0019_talkgroup_comments... OK
  Applying radio.0020_plan_profile... OK
  Applying radio.0021_default_plan... OK
  Applying radio.0022_system... OK
  Applying radio.0023_default_system... OK
  Applying radio.0024_auto_20170225_0857... OK
  Applying radio.0025_unit_slug... OK
  Applying radio.0026_auto_20170305_1336... OK
  Applying radio.0027_auto_20170306_2118... OK
  Applying radio.0028_auto_20170307_2152... OK
  Applying radio.0029_scanlist_slug... OK
  Applying radio.0030_scanlist_build_slug... OK
  Applying radio.0031_webhtml... OK
  Applying radio.0032_default_webhtml... OK
  Applying radio.0033_auto_20170514_2026... OK
  Applying radio.0034_auto_20170516_1852... OK
[... 49 lines cut ...]
    updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 903, in _save_table
    forced_update)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 953, in _do_update
    return filtered._update(values) > 0
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/query.py", line 662, in _update
    return query.get_compiler(self.db).execute_sql(CURSOR)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1191, in execut                                                                                                                     e_sql
    cursor = super(SQLUpdateCompiler, self).execute_sql(result_type)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 886, in execute                                                                                                                     _sql
    raise original_exception
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 876, in execute                                                                                                                     _sql
    cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/utils.py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/utils.py", line 65, in execute
    return self.cursor.execute(sql, params)
django.db.utils.IntegrityError: duplicate key value violates unique constraint "auth_permission_content_type_id_codename                                                                                                                     _01ab375a_uniq"
DETAIL:  Key (content_type_id, codename)=(27, add_emailaddress) already exists.

```

MaxwellDPS, 2017-11-17:
After commenting those lines out I got this
```
(Trunk Player)radio@FOX3:~/trunk-player$ ./manage.py move_db
Moving data from app radio table plan into new db
Moving data from app auth table user into new db
Moving data from app auth table user_groups into new db
Moving data from app auth table user_user_permissions into new db
Moving data from app account table emailaddress into new db
Moving data from app account table emailconfirmation into new db
Moving data from app sites table site into new db
Moving data from app radio table system into new db
Moving data from app radio table agency into new db
Moving data from app radio table source into new db
Moving data from app radio table talkgroup into new db
Traceback (most recent call last):
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/base/base.py", line 236, in _commit
    return self.connection.commit()
psycopg2.IntegrityError: insert or update on table "radio_talkgroup" violates foreign key constraint "radio_talkgroup__h                                                                                                                     ome_site_id_9c0b20ee_fk_radio_repeatersite_id"
DETAIL:  Key (_home_site_id)=(1) is not present in table "radio_repeatersite".

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 363, in execu                                                                                                                     te_from_command_line
    utility.execute()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/__init__.py", line 355, in execu                                                                                                                     te
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 283, in run_from_                                                                                                                     argv
    self.execute(*args, **cmd_options)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/core/management/base.py", line 330, in execute
    output = self.handle(*args, **options)
  File "/home/radio/trunk-player/radio/management/commands/move_db.py", line 54, in handle
    move_all_db_data(options)
  File "/home/radio/trunk-player/radio/management/commands/move_db.py", line 87, in move_all_db_data
    rec.save(using='default')
  File "/home/radio/trunk-player/radio/models.py", line 115, in save
    super(TalkGroup, self).save(*args, **kwargs)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 806, in save
    force_update=force_update, update_fields=update_fields)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/models/base.py", line 836, in save_base
    updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/transaction.py", line 223, in __exit__
    connection.commit()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/base/base.py", line 262, in commit
    self._commit()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/base/base.py", line 236, in _commit
    return self.connection.commit()
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/utils.py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)
  File "/home/radio/trunk-player/env/lib/python3.5/site-packages/django/db/backends/base/base.py", line 236, in _commit
    return self.connection.commit()
django.db.utils.IntegrityError: insert or update on table "radio_talkgroup" violates foreign key constraint "radio_talkg                                                                                                                     roup__home_site_id_9c0b20ee_fk_radio_repeatersite_id"
DETAIL:  Key (_home_site_id)=(1) is not present in table "radio_repeatersite".

(Trunk Player)radio@FOX3:~/trunk-player$
```

dreinhold, 2017-11-17:
how big is your sqlite db. Could I copy it off your machine so I can test
with it? - you can email me dreinhold@gmail.com

MaxwellDPS, 2017-11-17:
sure thing!

dreinhold, 2017-11-19:
Should be good after commit 20ab79a on master..

closed 2017-11-19 by dreinhold