ScanOC/trunk-player #50: list_pending_transmisions does not use redis info from settings

dreinhold, 2018-06-02:


EricTendian, 2018-09-04:
@dreinhold which particular setting should it ideally use the redis info from? It looks like redis config is defined in a couple places. I'm working on making this application compatible w/ Docker, here's what I've got for this particular item: https://github.com/ScanOC/trunk-player/compare/master...CrimeIsDown:docker-compose#diff-4fbcae94d259598edab53050d48ecbefR8

dreinhold, 2018-09-04:
I though I had prototyped this, but don't see the code.

The way you did it seems fine for now, some changes need to made to make it a little more customization.

One thing is we can follow the channels setup (which by default is probably the best way for now)

channel_redis_url = settings.CHANNEL_LAYERS['default']['CONFIG']['hosts'][0]
self.__db = redis.Redis.from_url(os.environ.get('REDIS_URL', channel_redis_url), **redis_kwargs)

kcwebby, 2020-09-21:
Dylan -- Any way we could get this wrapped in so it functions?
I'm trying to move away from SSH connections, and would like the recorder machine to add to a ElasticCache redis server, and then have the TP pick it up from there with the add_transmission_worker.

I am picking this issue up from where Rachael left it off, but unfortunately don't have the coding skills that she did.

kcwebby, 2020-09-21:
Is it as simple as changing radio/utility.py ?
https://github.com/ScanOC/trunk-player/blob/99137b085d7a9d17818155b17d891fb5595399fa/radio/utility.py#L5

redis.Redis(host='hostname',port=port,password='password')  is the format expected, how do modify to pick that up from the config file, that part I don't know.

kcwebby, 2020-10-07:
Update from my PR and a full day of testing... I've reached my limit on what I know, but I did explore a few options:

I have a configuration box setup, configured with the above pull request incorporated, and with local redis-server running it works fine (although, because its not pointed at the same redis server, the queue reloaded on the websocket is affected).

If you stop the redis-server on the webserver, and instead rely on the remote redis server, you get the 400 error with this exception. 

``` 
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/client.py", line 666, in execute_command
    connection.send_command(*args)
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 609, in send_command
    self.send_packed_command(self.pack_command(*args))
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 584, in send_packed_command
    self.connect()
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 488, in connect
    raise ConnectionError(self._error_message(e))
redis.exceptions.ConnectionError: Error 111 connecting to localhost:6379. Connection refused.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 483, in connect
    sock = self._connect()
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 540, in _connect
    raise err
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 528, in _connect
    sock.connect(socket_address)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/daphne/http_protocol.py", line 162, in process
    self.factory.channel_layer.send("http.request", {
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/asgi_redis/core.py", line 124, in send
    self.chansend(
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/client.py", line 2943, in __call__
    return client.evalsha(self.sha, len(keys), *args)
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/client.py", line 2078, in evalsha
    return self.execute_command('EVALSHA', sha, numkeys, *keys_and_args)
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/client.py", line 672, in execute_command
    connection.send_command(*args)
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 609, in send_command
    self.send_packed_command(self.pack_command(*args))
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 584, in send_packed_command
    self.connect()
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 488, in connect
    raise ConnectionError(self._error_message(e))
redis.exceptions.ConnectionError: Error 111 connecting to localhost:6379. Connection refused.

2020-10-06 20:12:50,014 Traceback (most recent call last):
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 483, in connect
    sock = self._connect()
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 540, in _connect
    raise err
  File "/home/ubuntu/trunk-player/env/lib/python3.8/site-packages/redis/connection.py", line 528, in _connect
    sock.connect(socket_address)
ConnectionRefusedError: [Errno 111] Connection refused

```

I attempted to locate / hard code for testing my redis server (on AWS) into the following locations with no change in behavior from the app:

`` ~/trunk-player/env/lib/python3.8/site-packages/redis/client.py:488 ``
`` ~/trunk-player/env/lib/python3.8/site-packages/redis/connection.py:437 ``

I also attempted to change the redis channel, and cache "default" keys in ~/trunk-player/trunk_player/settings.py to default00, since it also points to localhost, and the server would not start, so I hardcoded that to my AWS instance as well, with no change.

Any help would be appreciated on this.

closed 2024-03-21 by dreinhold