ScanOC/trunk-player #53: User accounts aren't being updated when they subscribe to paid plans

gavinrozzi, 2018-09-12:
Currently experiencing this problem with my instance of trunk-player. I have created a few different paid plans (https://ocradio.live/plans/), and users have been able to sign up for them without issue.

Recently, all new subscriptions have not been updating the user accounts of paid subscribers. While they show up just fine as customers in the administrative interface, their accounts are not being updated to reflect the additional level of access. For example, someone signs up for a plan that allows 24-hours access, yet their account is still showing as "free" despite the charge successfully going through and showing up on Stripe. This has required me to manually go in and update their accounts.

dreinhold, 2018-09-12:
In stripe do you have pending web hooks? Did your domain change, Try a test web hook from stripe.

gavinrozzi, 2018-09-13:
Ah! Looks like that was the problem after all. Thanks for your help.

closed 2018-09-13 by gavinrozzi