ScanOC/trunk-player #54: docker support?

jp83, 2018-11-07:
I'm not familiar with django apps and the architecture here, but have you thought using docker to make this easier to deploy?  Just curious if there's any particular reasoning, maybe eventually I might take a stab at it.

In particular, I'm finding (I think) that the ./encode-local-sys-0.sh method of adding transmissions isn't compatible with the dockerized version of recorder, trunk-recorder-docker because it can't execute outside of it's container.  Curious on thoughts to better handle this integration?

dreinhold, 2018-11-09:
I have never used docker, but would be happy to see it work in docker.
I don't understand the limitations, might need to make the encode/import process web based. 
The newer add_transmission_worker.py method might be made to work. As its just looking for messages in a queue which can be network based.

Gibby, 2019-02-15:
Took me a couple hours but I have this almost working in docker. With 3 containers(postgres,player,recorder)

I have the recorder doing the enode and moving files between shared volumes to the player.
I have the recorder using redis-cli to add the transmission.

Have to finish reading docs to figure out how to import talk groups and stuff.

Only thing in terms of the containers not be 100% automated is the createsuperuser part yet.

There are no docs for how to use it and it uses the fork on trunk-recorder which has my config.json and my csv already in.

Feel free to take a look at it here:
https://github.com/Gibby/trunk-player

Gibby, 2019-02-21:
I have got a working container that works with local_audio files.
https://github.com/Gibby/sdr-trunk

It includes trunk-recorder, trunk-player, and liquidsoap.