ScanOC/trunk-player #59: Scan lists do not auto update

frameshift18, 2019-01-13:
The scan lists pages do not auto update the incoming transmissions.
Uncommenting `setInterval(buildpage, 2000);`  does make it auto update though.

Not sure what the downside of uncommenting the above line has?
Curious if this other people have this issue, I haven't seen this brought up before.

frameshift18, 2019-01-13:
Hmm okay, any ideas how I could start to troubleshoot the websocket connection? Nginx config? Permissions?

dreinhold, 2019-01-13:
Start with the ./manage.py runworker make sure you have a couple running, then make sure you have reddis up and running, I think it might not even start daphnie if reddis is not running so that might be okay. In your browser bring up the development tools and console, re-fresh the page see if you get any errors about not being able to connect.

frameshift18, 2019-01-13:
So it looks like only the default scan list does not auto update.  If I make a scan list through the admin portal with all the currently known talk groups it does auto update like I would expect.
No web socket errors in the console.

I did have some web socket unable to connect errors through my reverse proxy but i added some lines to nginx config and now that is working, just no default scan list updating through that either.

dkettmann, 2019-01-13:
I have the same issue on my install. I kind of thought it might be because I somehow ended up deleting the "default" scan list from admin portal and rebuilt it by just throwing all channels in a list named similar.  Still might be, if you did this also.

But the actual built-in default list doesn't update when browsed to from the intro screen to enter the system, or the /scan/default.  Correction, it updated after about 4-5 minutes while I was typing this. But now nowhere like the ones I defined update. Seems to be updating only every few minutes.

frameshift18, 2019-01-13:
In troubleshooting this I did create a scan list named default and added all talk groups to it and that did update normally. Once i deleted that, the built-in default scan list started not updating again.

There's a js function to update the list on window focus, ie when you click off the browser and back to it, that does work on the built-in default list, that might be what you are seeing and didn't realize.

dreinhold, 2019-01-13:
I just pushed a fix for the default scan list. I have always overwritten the default scan-list (to limit what is in default, so I had not really noticed).

frameshift18, 2019-01-13:
Awesome it is working great now.

Did not expect this quick of a response and fix! Thanks again.

dkettmann, 2019-01-19:
Is it possible this issue would effect custom-scan lists? I had a user note that his custom scan list stopped refreshing.  This is a newly discovered feature to us, and it has only happened once, but I thought of this thread and wondered if could have been overlooked, and if so possibly easily fixed. Thanks again. 

(Also for the record fs18 guessed exactly what made mine refresh above).

dreinhold, 2019-01-25:
@dkettmann yes similar issue. That code you uncommented just forces the data to reload every 2 seconds and update the table. Ill re open this as it needs to be fixed also.

frameshift18, 2019-01-27:
Just noticed this same behavior with a custom scan list as well. I have been using them more and I seem to have had this issue only once so far.

kcwebby, 2020-02-05:
I'm still experiencing this, even on AWS without a reverse proxy.   In the
clients browser console I see constant errors like this:

reconnecting-websocket.min.js:1 WebSocket connection to 'ws://
xxxxxxxx.com/ws-calls/' failed: WebSocket is closed before the connection
is established.

e850205, 2023-11-07:
This is also a problem if you "hold" on a single talkgroup

closed 2024-03-20 by dreinhold