ScanOC/trunk-player #62: csv format not the same as trunk-recorder

Gibby, 2019-02-21:
For import talk groups with a csv file, the same csv file  between trunk-player and trunk-recorder can not be used . Would be great to align them up.

From my testing so far:
* trunk-player uses the csv that can be download from radioreference.com
* trunk-recorder uses the copy/paste method from radioreference.com
* A couple of the columns are in a different location.
* trunk-recorder can have a header line, trunk-player can not

Would be great to support either way.

fiveiron, 2019-07-16:
Forgive my ignorance but I don't see a place in trunk-player where I can load a TG CSV file.  What am I missing?  Thanks.

MaxwellDPS, 2020-02-04:
just ran into it my self for future reference positions,5-7 have to be ints ie 0

Here is a working line :)

`60526,0,D,NDOT 7 MinEn,Department of Transportation,0,0,0,3`

kcwebby, 2020-02-04:
I fixed this a long time ago --  I never put in a pull request though.
Really the import/export scripts in TP need a good re-write to align with
the format that is on RR.
I'm several commits off now, so let me pull down a fresh copy and see if I
can re-create it to push it back up for Dylan, unless what if you have
already done it.

the import command for the TG is  ./manage.py import_talkgroups
filename.csv --system=<INT>

MaxwellDPS, 2020-02-04:
Nope I havent if you get a chance that would be great.

Gibby, 2020-02-18:
Been awhile since I had to touch my system that is running. What I did was create a modified version of import_talkgroups.py that I was able to use with the format from RR. 

Here is my modified copy:
https://github.com/Gibby/sdr-trunk/blob/master/src_files/import_talkgroups2.py

I haven't pulled from RR for a while and I think there might have been a format change since I created import_talkgroups2.py