ScanOC/trunk-player #65: Audio playback issues

chrkyl, 2019-06-02:
I am new to trunk player and just installed it.  I think I have everything running, I am seeing new calls come in the site but when I try and play them, it doesn't work.  Looking at the terminal it is requesting it but it is saying the file is not found and displaying a path, however the path does match the location of the file.  I'm stumped.  I am sure I probably missed a step in the setup somewhere but I have gone back over it numerous times and cant find it.

dreinhold, 2019-06-03:
Where are your audio files being saved to/served from?
There a few ways to setup the audio files.

chrkyl, 2019-06-03:
Locally.  They are coming from trunk recorder going to /home/ckyle/radio/trunk-player/audio_files/.

chrkyl, 2019-06-04:
This is what i get in the terminal:
2019-06-04 12:35:12,975 Not Found: /audio_files/sys_1/2019/6/4/2-1559666100_8.57438e+08.mp3
[2019/06/04 12:35:13] HTTP GET /audio_files/sys_1/2019/6/4/2-1559666100_8.57438e+08.mp3 404 [0.06, 127.0.0.1:38932]

dkettmann, 2019-06-04:
Its been a long time but I do remember a small struggle to get the audio_files directory working right.  Looking at my notes (specifically part 3) I posted on the google group ( https://groups.google.com/forum/#!topic/trunk-player/CR70pVzA6pk ), it was either in settings_local.py or the trunk_player.nginx config file that was wrong.  Wish I could remember better but maybe this points in the right direction?  I think it was a relative path that was needed and not the full path but I might be off on that.  Maybe this helps a bit?

dreinhold, 2019-06-05:
Do you have nginx running the site. It looks like the audio_file setting is not working. It should serve the files. django should not see that request.

chrkyl, 2019-06-06:
Got it.  I think it was an issue with nginx.  The setting file wasnt linked correctly.  Its working now.  Thanks!

closed 2019-06-06 by chrkyl

e850205, 2020-07-23:
I am having this issue. Does anyone have any advice?