ScanOC/trunk-player #72: Ubuntu 18.04.3

kcwebby, 2019-09-02:
I seemed to have broken things when I did a distro release update.

Now all I'm getting is this error, pointing to a python error.

Doing a google search, I tried setting an environment variable, and that got me further, but still didn't solve the problem.

Anyone seen this?  (thats what I get for upgrading I guess)

(MSTP)radio@scanweb1:/home/radio/trunk-player$ export PYTHONPATH=$PYTHONPATH:/usr/lib/python3
(MSTP)radio@scanweb1:/home/radio/trunk-player$ ./manage.py runworker
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Failed to import the site module
Traceback (most recent call last):
  File "/usr/lib/python3.6/site.py", line 79, in <module>
    import os
  File "/usr/lib/python3.6/os.py", line 652, in <module>
    from _collections_abc import MutableMapping
  File "/usr/lib/python3.6/_collections_abc.py", line 64
    async def _ag(): yield
                    ^
SyntaxError: 'yield' inside async function

dreinhold, 2019-09-02:
> Consider setting $PYTHONHOME to [:<exec_prefix>]

Are you using a virtual environment? If so I found the easiest way is just
delete the env folder, re build the virtualenv, activate it and re pip
install the file modules from requirements.txt

kcwebby, 2019-09-02:
Yeah -- Here's the solution:

On a hunch, I commanded install of: python3-django-sitetree
and it informed me:
The following additional packages will be installed:
  javascript-common libjs-jquery python-django-common python3-django python3-sqlparse python3-tz

Then rebuilt the virtualenv (deleted the env/ dir and resetup)

Seems to be working now, I probably could have deleted and redid the virtualenv, it might have worked, but it was calling some missing python functions, so I'm not sure.

closed 2019-09-02 by kcwebby