ScanOC/trunk-player #79: json.decoder.JSONDecodeError: Extra data: line 1 column 14 (char 13)

mattmac553, 2020-06-27:
Hello,
I am trying to have trunk-recorder upload analog calls.
In trunk recorder I get this message every call, I have not been able to receive a single call yet. What do I need to do?

[2020-06-27 17:26:11.000526] (info)   [systn]	TG:          6 (               Fire 1)	Freq: 5.068375e+08	Ending Recorded Call - Last Update: 47s	Call Elapsed: 47
[2020-06-27 17:26:11.001888] (info)   Running upload script: ./encode-local-sys-0.sh /Users/matthewcarango/trunk-player/audio_files/delco/2020/6/27/6-1593293124_506837500.wav &
Encoding: /Users/matthewcarango/trunk-player/audio_files/systn/2020/6/27/6-1593293124_506837500.wav
head: illegal line count -- -2
LAME 3.100 64bits (http://lame.sf.net)
polyphase lowpass filter disabled
Encoding /Users/matthewcarango/trunk-player/audio_files/systn/2020/6/27/6-1593293124_506837500.wav
      to /Users/matthewcarango/trunk-player/audio_files/systn/2020/6/27/6-1593293124_506837500.mp3
Encoding as 8 kHz single-ch MPEG-2.5 Layer III (2.3x) average 56 kbps qval=3
    Frame          |  CPU time/estim | REAL time/estim | play/CPU |    ETA 
   487/487   (100%)|    0:00/    0:00|    0:00/    0:00|   549.82x|    0:00 
  8 [ 20] *********
 16 [  0] 
 24 [  1] *
 32 [  2] *
 40 [  0] 
 48 [ 90] *****************************************
 56 [369] *********************************************************************************************************************************************************************
 64 [  5] ***
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
   kbps       mono %     long switch short %
   52.5      100.0        68.0  17.7  14.4
Writing LAME Tag...done
ReplayGain: -3.6dB
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/Users/matthewcarango/trunk-player/env/lib/python3.8/site-packages/django/core/management/__init__.py", line 364, in execute_from_command_line
    utility.execute()
  File "/Users/matthewcarango/trunk-player/env/lib/python3.8/site-packages/django/core/management/__init__.py", line 356, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/Users/matthewcarango/trunk-player/env/lib/python3.8/site-packages/django/core/management/base.py", line 283, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/Users/matthewcarango/trunk-player/env/lib/python3.8/site-packages/django/core/management/base.py", line 330, in execute
    output = self.handle(*args, **options)
  File "/Users/matthewcarango/trunk-player/radio/management/commands/add_transmission.py", line 60, in handle
    add_new_trans(options)
  File "/Users/matthewcarango/trunk-player/radio/management/commands/add_transmission.py", line 135, in add_new_trans
    data = json.load(data_file)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/json/__init__.py", line 293, in load
    return loads(fp.read(),
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/json/decoder.py", line 340, in decode
    raise JSONDecodeError("Extra data", s, end)
json.decoder.JSONDecodeError: Extra data: line 1 column 14 (char 13)

mattmac553, 2020-06-27:
I am using a Mac

dreinhold, 2020-06-27:
It does not like the json file from trunkrecorder, can you post one of the json files that is being created.

mattmac553, 2020-06-27:
"play_length": 0.840625,
"source": 0,
"freqList": [ { "freq": 508237500.000000, "time": 1593295233, "pos": 0.000000, "len": 0.000000, "error_count": 0.000000, "spike_count": 0.000000}]
}
@dreinhold

dreinhold, 2020-06-27:
There should be more in that file.
Does your encode upload have these lines in it?
https://github.com/ScanOC/trunk-player/blob/master/utility/trunk-recoder/encode-upload.sh#L21-L24

mattmac553, 2020-06-28:
Thats all it says since I have installed Trunk-Player.   Before it would say but not always all of that source if not every transmission would have it:

{
"freq": 5.06588e+08,
"start_time": 1593134156,
"stop_time": 1593134221,
"emergency": 0,
"talkgroup": 2,
"srcList": [ {"src": 38020, "time": 1593132746, "pos": 0.247750, "emergency": 0, "signal_system": "MDC1200", "tag": ""}, {"src": 38017, "time": 1593133498, "pos": 0.267375, "emergency": 0, "signal_system": "MDC1200", "tag": ""}, {"src": 38006, "time": 1593134099, "pos": 0.258500, "emergency": 0, "signal_system": "MDC1200", "tag": ""}, {"src": 30608, "time": 1593134208, "pos": 0.274875, "emergency": 0, "signal_system": "MDC1200", "tag": ""} ],
"freqList": [ ]
}

mattmac553, 2020-06-28:
Correction right now the program seems to be deleting the files right after they pop up.
@dreinhold

MaxwellDPS, 2021-10-08:
@mattmac553 Is this issue still persisting?

closed 2024-03-20 by dreinhold