ScanOC/trunk-player #86: Djangorestframework > 3.11 Incompatible with Django 1.11, breaks build

sally-yachts, 2020-10-17:
Issue replicated on freshly deployed and updated Ubuntu 20.04 as well as 18.04.

When installing requirements via pip, the following error is seen:

`djangorestframework 3.12.1 requires django>=2.2, but you'll have django 1.11.29 which is incompatible.`

If this error isn't noticed and installation process is continued, then when attempting to run `./manage.py migrate` the following error is also seen:

`ImportError: cannot import name 're_path' from 'django.urls' (/opt/trunk-player/env/lib/python3.8/site-packages/django/urls/__init__.py)`

Per the release notes for djangorestframework, the 3.11 branch is the last to support django 1.11. I changed the appropriate line in requirements.txt to the following and the situation was resolved:

`djangorestframework<3.12 `

That's the only package that uses greater than as a requirement selector, so the only other issues that may eventually pop up is if one of the packages that don't have any requirement selectors defined eventually drops support for django 1.11.

bctrainers, 2020-10-18:
You can safely upgrade djangorestframework to version 3.11.1. After 3.11.1, things will be broken. :)

sally-yachts, 2020-10-18:
> You can safely upgrade djangorestframework to version 3.11.1. After 3.11.1, things will be broken. :)

Just to be clear, do you mean you noticed issues on 3.11.2 or are you meaning to confirm the issue that 3.12 is incompatible? If there's issues with 3.11.2 then I can update the PR and change the setting to allow up to 3.11.1.

bctrainers, 2020-10-18:
My apologies, I wasn't aware djangorestframework had been updated to 3.11.2 on Sept 30th. I legitimately updated my packages one by one the week prior on my testing environment. 

I would assume that the 3.11.2 version is functional without issue. 3.11.1 is what I have right now, and there are no issues.

I will dig into 3.12.x later this weekend, as there appear to be some rather notable changes.

sally-yachts, 2020-10-18:
No worries, just making sure! I noticed the recent release date as I started poking at this issue while prepping a fresh VM, didn't want to accidentally suggest a bad version in the PR.

dreinhold, 2020-10-18:
Cool thanks for the PR. Yea this needs to get off django 1.x. but the main holdup the django-channel's web-sockets, that part of trunkplayer needs to be re-written compltly to move django up at all.

closed 2020-10-18 by dreinhold

kritik-Trellis, 2022-01-11:
akismet==1.1
amqp==1.4.9
anyjson==0.3.3
APScheduler==3.8.1
arabic-reshaper==2.1.3
asgi-redis==1.4.3
asgiref==1.1.2
attrs==21.4.0
autobahn==21.11.1
Automat==20.2.0
backcall==0.2.0
backports.functools-lru-cache==1.6.4
backports.shutil-get-terminal-size==1.0.0
backports.zoneinfo==0.2.1
beautifulsoup4==4.10.0
billiard==3.3.0.23
cached-property==1.5.2
caniusepython3==7.3.0
celery==3.1.17
certifi==2021.10.8
cffi==1.15.0
channels==1.1.2
chardet==4.0.0
charset-normalizer==2.0.10
click==8.0.3
click-didyoumean==0.3.0
click-plugins==1.1.1
click-repl==0.2.0
constantly==15.1.0
coreapi==2.3.3
coreschema==0.0.4
cryptography==36.0.1
cycler==0.11.0
daphne==1.4.2
decorator==4.3.0
defusedxml==0.7.1
Deprecated==1.2.13
distlib==0.3.4
Django==1.11.17
django-admin-honeypot==1.1.0
django-allauth==0.26.1
django-appconf==1.0.5
django-bootstrap-form==3.4
django-celery==3.1.17
django-compat==1.0.15
django-compressor==3.1
django-constance==2.8.0
django-extensions==2.1.6
django-filter==1.1.0
django-helpdesk==0.2.10
django-hijack==2.1.5
django-hijack-admin==2.1.10
django-js-asset==1.2.2
django-markdown-deux==1.0.5
django-modeladmin-reorder==0.3.1
django-mptt==0.13.4
django-otp==0.7.5
django-picklefield==2.0
django-pyodbc-azure==1.11.13.1
django-rest-swagger==2.2.0
django-richtextfield==1.6.1
django-simple-history==3.0.0
django-taggit==0.23.0
djangorestframework==3.11.1
email-reply-parser==0.5.12
enum34==1.1.10
et-xmlfile==1.1.0
fonttools==4.28.5
future==0.18.2
fuzzywuzzy==0.18.0
html5lib==1.1
httpagentparser==1.9.1
hyperlink==21.0.0
idna==3.3
imageio==2.13.5
imageio-ffmpeg==0.4.5
imgkit==1.2.2
importlib-metadata==4.10.0
incremental==21.3.0
install==1.3.5
ipython==7.31.0
ipython-genutils==0.2.0
itypes==1.2.0
jedi==0.18.1
Jinja2==3.0.3
joblib==1.1.0
jsonfield==2.0.2
kiwisolver==1.3.2
kombu==3.0.37
logzero==1.7.0
Louie==2.0
lxml==4.7.1
Markdown==3.3.6
markdown2==2.4.2
MarkupSafe==2.0.1
matplotlib==3.5.1
matplotlib-inline==0.1.3
moviepy==1.0.3
msgpack-python==0.5.6
nltk==3.6.7
numpy==1.21.5
oauthlib==3.1.1
openapi-codec==1.3.2
openpyxl==3.0.9
packaging==21.3
pandas==1.3.5
parso==0.8.3
pathlib==1.0.1
pathlib2==2.3.6
pexpect==4.8.0
pickleshare==0.7.5
Pillow==9.0.0
pretty-cron==1.2.0
proglog==0.1.9
prompt-toolkit==3.0.24
ptyprocess==0.7.0
pur==5.4.0
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycparser==2.21
pycrypto==2.6.1
pycryptodome==3.12.0
Pygments==2.11.1
PyJWT==2.3.0
pymssql==2.2.3
pyodbc==4.0.23
pyOpenSSL==21.0.0
pyparsing==3.0.6
PyPDF2==1.26.0
python-bidi==0.4.2
python-crontab==2.6.0
python-dateutil==2.8.2
python-keyczar==0.716
python-openid==2.2.5
python-slugify==5.0.2
python3-openid==3.2.0
pytz==2021.3
pytz-deprecation-shim==0.1.0.post0
PyYAML==6.0
pyzmq==22.3.0
rcssmin==1.1.0
redis==2.10.6
regex==2021.11.10
reportlab==3.6.5
requests==2.27.1
requests-oauthlib==1.3.0
rjsmin==1.2.0
scandir==1.10.0
scikit-learn==1.0.2
scipy==1.7.3
service-identity==21.1.0
simplegeneric==0.8.1
simplejson==3.17.6
six==1.16.0
sklearn==0.0
soupsieve==2.3.1
sqlparams==3.0.0
sqlparse==0.4.2
text-unidecode==1.3
textblob==0.17.1
threadpoolctl==3.0.0
tqdm==4.62.3
traitlets==5.1.1
twilio==7.4.0
Twisted==21.7.0
txaio==21.2.1
typing_extensions==4.0.1
tzdata==2021.5
tzlocal==4.1
uritemplate==4.1.1
urllib3==1.26.7
vaderSentiment==3.3.2
vine==5.0.0
wcwidth==0.2.5
webencodings==0.5.1
wrapt==1.13.3
xhtml2pdf==0.2.5
xlrd==2.0.1
XlsxWriter==3.0.2
zipcodes==1.2.0
zipp==3.7.0
zope.interface==5.4.0

I have installed Django =1.11.17 and djaangorestframework==3.11.1
still I am getting error - 
**File "/home/kritik/Trellissoft/empereon/empereon_py3_django1/lib/python3.7/site-packages/simple_history/admin.py", line 9, in <module>
    from django.urls import re_path, reverse
ImportError: cannot import name 're_path' from 'django.urls' (/home/kritik/Trellissoft/empereon/empereon_py3_django1/lib/python3.7/site-packages/django/urls/__init__.py)**

Can anyone help me to resolve this ?

MaxwellDPS, 2022-01-14:
@kritik-Trellis What version are you running?