ScanOC/trunk-player #88: Feature: Unit Description Searchable by desc ID on scan page

s3m1s0n1c, 2020-12-22:
Hey Guys,

So I was thinking it would be awesome if the Unit ID Description on the scan page was a clickable link.
When the user clicks the link it will do a /unit/1234567 search for all units of that unit id.

Example: If i click NorthernOps-GosfordDispatch it will look up the dec id and use it in site.com/unit/1234567 so one can search all calls for that unit description.

[image]

Thanks
Sonic

frameshift18, 2020-12-22:
I also was wanting this feature and I saw it on this commit https://github.com/gopher2/trunk-player/commit/07aa2d4bb7b97ff2bc38e0c5e71bc0741f1e6ec4

I've been using it for a week and it seems to work well. One thing to note is if an unit id has a description it will search by the description and not the unit id when you click on it. ie: /unit/{description}

s3m1s0n1c, 2020-12-22:
@frameshift18 Ahh cool.. I didn't see this commit from gopher2

I have tried it but cant get it to work.... It seems no changes where made..

I have rebooted, restarted services and did manage.py collectstatic and no changes to page.

Have i missed something obvious. Im in holiday brain mode.

Thanks
Sonic

frameshift18, 2020-12-22:
Changes to the .js files should be reflected after you do

$ source env/bin/activate
$ ./manage.py collectstatic --noinput

If you don't see any changes the trunkplayer.js file may be cached by your browser.

s3m1s0n1c, 2020-12-22:
Ok All good Thanks..

I've commented on gophers commit

dreinhold, 2020-12-22:
If @￼gopher2 want to send a pull request we can merge it in with a few small changes.
I would warn is you have a large amount of data the search by unit can be slow, might be best to have it as an option to can enable or disable.

frameshift18, 2020-12-22:
@s3m1s0n1c you shouldn't have to change any of your descriptions to use that commit to do what you want. If you click on a unit description you should be able to search by that. It would just use the slug. But that doesn't matter since it all points to the same unit. Am I missing something?

The only negative about this is that since /unit/ uses the slug, and since the slug would be changed to the description you loose the ability to /unit/{dec_id} on a unit id that has a description set. This could be changed by adding one line to views.py though.

frameshift18, 2020-12-22:
Oh I think I see what's going on....
Since this commit uses the unit description in the html link, if there's a space in the description then it doesn't work correctly because spaces are dashes in the slug. I never noticed this since I was only messing around with one word descriptions lol...

frameshift18, 2020-12-22:
@s3m1s0n1c Check this commit and see how that works for you. The basics are there. https://github.com/frameshift18/trunk-player/commit/fdd4e590d3673769b57a242d30a84db686d2c560

s3m1s0n1c, 2020-12-22:
> @s3m1s0n1c Check this commit and see how that works for you. The basics are there. [frameshift18@fdd4e59](https://github.com/frameshift18/trunk-player/commit/fdd4e590d3673769b57a242d30a84db686d2c560)

:) Thanks for that.. I'll give it a whirl soon.. Waiting for my brain to work again.....

s3m1s0n1c, 2020-12-22:
@frameshift18
Thank you so much... That works a treat...

It only works for me logged in as admin tho?

Is there A way i can make it work for a specific plan name or plan number?

Thanks
sonic

s3m1s0n1c, 2020-12-22:
Guess I wrap it with a {% if user.is_authenticated %}?

frameshift18, 2020-12-22:
It's because that edit link takes you to the admin edit page of that unit id. So the user will need at least 'Staff status' and the 'radio | unit | Can change unit' permission. Like you stated earlier its probably a better idea to just link it back to the update unit modal as it was originally. I like the admin page because it allows more information to be seen/edited than just the description but it doesn't flow nearly as well. I might have a crack at that too.

RadioTechGuy, 2020-12-22:
I don't think you would need to give them staff access. just add the [Can change unit] access.

frameshift18, 2020-12-22:
With the original unit change modal yes you just need that permission. But since in my commit it links back to the admin site to edit the unit information in my quick testing you need Staff status to allow you into the admin site.

RadioTechGuy, 2020-12-22:
I did not notice the link back into admin. Yes, you would need staff for that.

s3m1s0n1c, 2020-12-22:
Mmm don't really wanna give members staff access just to be able to have a unit description searchable.. its less then ideal..

Any other ways around it to give members access?

s3m1s0n1c, 2020-12-22:
So it works if I give the members 'radio | unit | Can change unit' permission and delete the edit bit not they can edit anyways cause they dont have access..

I guess that will have to do for now..

Thanks guys for ya help :)

s3m1s0n1c, 2020-12-22:
Another issue I see with this way is...
Comms Rooms sometimes have different Radio ID's for the same thing..

Eg.. FireControl has 2 Radio Unit Entries as 12345 & 67890

They both show as FireControl on the page but If I click the link then its going to one or the other..

If u can search unit ID by slug as the slug would be the same..

Thanks

frameshift18, 2020-12-22:
> So it works if I give the members 'radio | unit | Can change unit' permission and delete the edit bit not they can edit anyways cause they dont have access..
> 
> I guess that will have to do for now..
> 
> Thanks guys for ya help :)

Yeah that's why I think it would be better to just use the original method with the unit edit modal box that pops up.

You can give staff access and radio change access but the only thing they will have access to in the admin portal is the unit edit screen.

The same unit name with different unit ids should work correctly as the search and edit are based off the unit id.

Oh.. I think I understand what you are saying. You would like the search page to show the transmissions for both dec_ids because a single 'user' can use multiple radio ids. You can use a '+' in the /unit/ search to search for multiple slugs or dec_ids now that I think about it. ie: '/unit/12345+67890'

Could just possibly go back to using the slug if there's a description set.
Let me think about it and whip something up. I'm sure I'm over thinking this. haha.

frameshift18, 2020-12-23:
@s3m1s0n1c Give this one a try.. https://github.com/frameshift18/trunk-player/commit/9a76800e308ebc2a3cf6360176a47754937bfb63

s3m1s0n1c, 2020-12-23:
@frameshift18 LEGEND!!!

Yep that seems to work..  Will keep testing..

Thanks man 👍

s3m1s0n1c, 2020-12-23:
So a few little issues..

This commit works with slugs and works with ones without a slug in the way of "Unit Searching".. 
But it allows all logged in users to change the unit id..
This is bad as I feel some users are just gonna put rubbish in there and totally screw up my hard work naming them.

Is there a way to make it so there is a ``js_config.radio_view_unit)`` or use ``js_config.radio_change_unit`` to just view and ``js_config.radio_add_unit`` to use the updateunit modal?

I've tried the following and a few other things but it broke the code..

```if(data.results[a].units[unit].description) {
                  if(js_config.radio_change_unit) {
                          //new_html += '<a href="/unit/' + data.results[a].units[unit].dec_id + '/">' + data.results[a].units[unit].description + '</a>';
                                  new_html += '<a href="/unit/' + data.results[a].units[unit].slug + '/">' + data.results[a].units[unit].description + '</a>';
                  } else {
                      new_html += data.results[a].units[unit].description + ', ';
                  }
              } else {
                  if(js_config.radio_change_unit) {
                      new_html += '<a href="/unit/' + data.results[a].units[unit].dec_id + '/">' + data.results[a].units[unit].dec_id + '</a>';
                  } else {
                      new_html += data.results[a].units[unit].dec_id + ', ';
                  }
              } else {
                  if(js_config.radio_add_unit) {
                      new_html += '<a href="/unit/' + data.results[a].units[unit].dec_id + '/">' + data.results[a].units[unit].dec_id + '</a>';
                                          new_html += '<a href="/unitupdate/' + data.results[a].units[unit].pk + '/" data-toggle="modal" data-target="#unitupdatemodal">#</a>, ';
                  } else {
                      new_html += data.results[a].units[unit].dec_id + ', ';
                  }
              } else {
                  if(js_config.radio_add_unit) {
                      new_html += '<a href="/unit/' + data.results[a].units[unit].dec_id + '/">' + data.results[a].units[unit].dec_id + '</a>';
                                       new_html += '<a href="/unitupdate/' + data.results[a].units[unit].pk + '/" data-toggle="modal" data-target="#unitupdatemodal">#</a>, ';
                  } else {
                      new_html += data.results[a].units[unit].dec_id + ', ';
                  }

              }
          }

```

Thanks
Sonic

frameshift18, 2020-12-23:
If an user doesn't have the change unit permission it should just show the description or dec_id text and not have it as a link. This is working in my environment.

I was planning to mess around with the permissions to get some more granular control over showing the unit ids.

With all the if statements that modify `new_html` you have to be careful the html code is correct for each condition. You could potentially not close out a html tag etc, and may break rendering.

s3m1s0n1c, 2020-12-24:
I don't think you understood what I was saying..
I agree the Unit Searchable part works fine.. being a member vs anonymous

It would be nice if logged in user could have "Unit Searchable" function but not be able to edit the unit id as this could quickly screw all my unit id descriptions and allow a "Editors" group to edit them or even admin's.

That's why I thought if I could do the following.

Give editors/admin the "Add Unit ID" permission and they can edit them.
Give members the "Change Unit ID" permission and they cannot edit.

Unless someone wants to make a View Unit ID permission in the system which would solve all issues..

Where would one add this kinda function?

frameshift18, 2020-12-24:
Give this a try I think it's correct..

```
              if(data.results[a].units[unit].description) {
                  if(js_config.radio_add_unit && js_config.radio_change_unit) {
                      new_html += '<a href="/unit/' + data.results[a].units[unit].slug + '/">' + data.results[a].units[unit].description + '</a>';
                      new_html += '<a href="/unitupdate/' + data.results[a].units[unit].pk + '/" data-toggle="modal" data-target="#unitupdatemodal">#</a>, ';
                  } else if(js_config.radio_add_unit && !js_config.radio_change_unit) {
                      new_html += '<a href="/unit/' + data.results[a].units[unit].slug + '/">' + data.results[a].units[unit].description + '</a>, ';
                  } else {
                      new_html += data.results[a].units[unit].description + ', ';
                  }
              } else {
                  if(js_config.radio_add_unit && js_config.radio_change_unit) {
                      new_html += '<a href="/unit/' + data.results[a].units[unit].dec_id + '/">' + data.results[a].units[unit].dec_id + '</a>';
                      new_html += '<a href="/unitupdate/' + data.results[a].units[unit].pk + '/" data-toggle="modal" data-target="#unitupdatemodal">#</a>, ';
                  } else if(js_config.radio_add_unit && !js_config.radio_change_unit) {
                      new_html += '<a href="/unit/' + data.results[a].units[unit].dec_id + '/">' + data.results[a].units[unit].dec_id + '</a>, ';
                  } else {
                      new_html += data.results[a].units[unit].dec_id + ', ';
                  }
              }
```

You will have to add `js_json['radio_add_unit'] = user.has_perm('radio.add_unit')` to radio_js_config.py

Also I switched your logic so having the add_unit permission _only_, will allow clickable unit search. Having _both_ add_unit and change_unit will allow clickable search and the link to the modal edit unit description popup. Having the change_unit permission for clickable links would still let them change the unit if they knew the url to go to.

The correct way to do this would be to have a view permission. Which can be added manually with a migration but I swore Django had a default view permission but I can't seem to get it to work.

s3m1s0n1c, 2020-12-24:
@frameshift18 Oh you are awesome..
Works well..  Yeah normal users wouldn't really know /unit exists..  no worried if a few use it.. I guess i'll monitor it and see how it goes.. Worst come to worst just take the function away to unit search...

Ahhh dang on the view permission.. Oh well..

My NodeJS is a little rusty and trying to get my head around Django gives me a head

Thanks again for your help..

bctrainers, 2020-12-24:
This is what I've done on a TrunkPlayer install for ./radio/js/trunkplayer.js

```javascript
                    for (unit in data.results[a].units)
                    {
                        has_units = true;
                        if (data.results[a].units[unit].description)
                        {
                            if (js_config.radio_change_unit)
                            {
                                new_html += data.results[a].units[unit].description;
                                new_html += ' <a href="/unit/' + data.results[a].units[unit].dec_id + '/">[L]</a><a href="/admin/radio/unit/' + data.results[a].units[unit].pk + '/">[A]</a><a href="/unitupdate/' + data.results[a].units[unit].pk + '/" data-toggle="modal" data-target="#unitupdatemodal">[T]</a>, ';
                            }
                            else
                            {
                                new_html += data.results[a].units[unit].description + ', ';
                            }
                        }
                        else
                        {
                            if (js_config.radio_change_unit)
                            {
                                new_html += data.results[a].units[unit].dec_id;
                                new_html += ' <a href="/unit/' + data.results[a].units[unit].dec_id + '/">[L]</a><a href="/admin/radio/unit/' + data.results[a].units[unit].pk + '/">[A]</a><a href="/unitupdate/' + data.results[a].units[unit].pk + '/" data-toggle="modal" data-target="#unitupdatemodal">[T]</a>, ';
                            }
                            else
                            {
                                new_html += data.results[a].units[unit].dec_id + ', ';
                            }
                        }
                    }
```

Then, in ./radio/views.py I have this for the UnitFilterViewSet class...

```python
class UnitFilterViewSet(generics.ListAPIView):
    serializer_class = TransmissionSerializer

    def get_queryset(self):
        unit_var = self.kwargs['filter_val']
        search_unit = re.split('[\+]', unit_var)
        q = Q()
        for s_unit in search_unit:
            q |= Q(slug__iexact=s_unit)
            q |= Q(dec_id__iexact=s_unit)
        units = Unit.objects.filter(q)
        rc_data = Transmission.objects.filter(units__in=units).filter(talkgroup_info__public=True).prefetch_related('units').distinct()
        #rc_data = limit_transmission_history(self.request, rc_data)
        rc_data = limit_transmission_history_six_months(self.request, rc_data)
        restricted, rc_data = restrict_talkgroups(self.request, rc_data)
        return rc_data
```

s3m1s0n1c, 2020-12-25:
Nice @bctrainers

s3m1s0n1c, 2021-08-16:
Closing this solution was found from above posts.

Thanks Guys

s3m1s0n1c, 2022-11-09:
Hey @bctrainers & @frameshift18,
I cant get this working on the newer version of trunk-player..

Has anything changed?

Thanks
Sonic

bctrainers, 2022-11-17:
@s3m1s0n1c - nothing has changed, but I have not yet upgraded to the newest TrunkPlayer on the repo. There are likely differences in the code between Django versions.

closed 2024-03-20 by dreinhold