ScanOC/trunk-player #89: Feature: Registration - Auto Add to specific group

s3m1s0n1c, 2020-12-22:
Is it possible to add a user to a specific group once they confirm there email?

Its very time consuming adding users every time they register.. 

If we could automate this process it would be awesome :)

Thanks
Sonic

dreinhold, 2020-12-23:
The all-auth module does fire an event/signal when the user confirms their email
```
allauth.account.signals.email_confirmed(request, email_address)
Sent after the email address in the db was updated and set to confirmed.
```
That could be coded around. Right now the radio/receivers.py is setup for capturing signals from the pintax stuff. It could just be put right in there.
"Just thinking out loud"....

s3m1s0n1c, 2020-12-23:
Ooo.. I'm picking up what your putting down..

Will give it a red hot shot..

Thanks for your help.. its a starting point.

s3m1s0n1c, 2021-07-24:
Sorry I might need a little more instruction this..

What command do I use to add a user to a group and plan?

And I add it in radio/receivers.py?

Thanks
Sonic

s3m1s0n1c, 2021-07-24:
Im not using Pintax stuff tho.. 
A user registers and they get logged in..
There's no email confirmation

Thanks

s3m1s0n1c, 2021-07-24:
i've tried this
``` if new_user is not None:
                if new_user.is_active:
                    stripe_actions.customers.create(user=new_user)
                    user.save()
                    my_group = Group.objects.get(name='Members')
                    my_group.user_set.add(username)
                    login(request, new_user)
                    return HttpResponseRedirect('/scan/default/')```
and
```
if form.is_valid():
            user = User.objects.create_user(
            username=form.cleaned_data['username'],
            password=form.cleaned_data['password1'],
            email=form.cleaned_data['email']
            )
            username = form.cleaned_data['username']
            password = form.cleaned_data['password1']
            new_user = authenticate(username=username, password=password)
            group = Group.objects.get(name='Members')
            group.user_set.add(username)```
            
            Both failed

dreinhold, 2021-07-24:
I guess i am not clear on this one.

Are you looking to set a default plan or default group (I don't think tr does anything with the default group).
Or are you asking about the talkgroup access groups, if so there is a flag in those to set as the default for new users..

s3m1s0n1c, 2021-07-24:
Hey Dylan,

Sorry If I wasn't explaining it right..

I would like to add a new user to the following after they register.

* Authentication and Authorization -> Group - Control Permissions(What they can access etc)

[image]

* Radio -> Plans - Control how far they can go back in time for transmissions

[image]

Hope this clears up any confusion..

Thanks in advance for your help and time.

Thanks
Sonic

dreinhold, 2021-07-24:
The plan stuff, if you go into the plan there is a check box to make the plan the default. That one should get set for new users.

As for the Group stuff, try and add this to the end of receivers.py 

```
def set_user_group(sender, instance, created, **kwargs):
    logger.error('Saved User')
    default_group, created = Group.objects.get_or_create(name='Member')
    default_group.user_set.add(instance)

post_save.connect(set_user_group, sender=User, dispatch_uid="set_user_default_group")

```
At the top edit the line
`from django.contrib.auth.models import User`
to
`from django.contrib.auth.models import Group, User`

s3m1s0n1c, 2021-07-25:
Yeah unfortunately, I was unable to get any of that to work by adding that code.
:(

Will keep chipping away @ it..

Thanks heaps for your help so far :)

s3m1s0n1c, 2021-07-28:
@dreinhold 
Maybe a easier answer is can I add "Member" as a X or Tick on the Users List (admin/auth/user/)
That way I can sort by Member and just go thru them and add them manually every day..

Example Below

[image]

Thanks
Sonic

s3m1s0n1c, 2021-08-06:
So I ended up getting this working after alot of trial and errors.

For those playing @ home....

Edited ```trunk-player/radio/receivers.py``` and added this to the end
```
@receiver(post_save, sender=User)
def create_user_profile(sender, instance, created, **kwargs):
    if created:
        instance.groups.add(Group.objects.get(name='group_name'))  
        
 ```

closed 2021-08-06 by s3m1s0n1c