ScanOC/trunk-player #93: BUG: Mobile Phone doesn't keep playing when screen asleep/timesout

s3m1s0n1c, 2020-12-28:
Hey Guys,

Looks like trunk-player doesn't keep playing on mobiles when the screen times out or goes to sleep...
It plays the call then stops and doesn't keep playing the next call..

Is there an easy way to fix this as my users would like to have it in the BG so they can hear whats going on.

Thanks
Sonic

s3m1s0n1c, 2020-12-28:
After doing some tests..

Safari is the only one at the moment..

Doing more tests on other browsers

bctrainers, 2020-12-28:
Android as a whole will do that sort of behavior. Once the screen timeout happens, Android will begin to shush/put-to-sleep apps that use the most CPU time thus most battery usage.

The only real way around that issue is by using an app (ScannerRadio, VLC, MXPlayer), Tasker or some other sort of keep-alive application, or having users go to the developer settings on android to force the phone screen to always remain on while charging.

s3m1s0n1c, 2020-12-28:
So it seems its not just mobiles.. 
If I open Chrome on my Mac and not actively on the table it stops playing after a while..
Its doing the same on mobile as well..

bctrainers, 2020-12-28:
Correct. Background tabs are throttled on Chrome (and most Chromium-based browsers) - timer dependent.

There used to be an option under chrome://flags to override the background tab CPU limiter. It would appear it was changed in recent years to: `chrome://flags/#intensive-wake-up-throttling`

dreinhold, 2020-12-28:
Yeah the browser have locked down a lot of javascript activities when the page is not in focus.
I'm sure there are some workarounds but probably each browser has their own.
The mplayer that was added 4 years ago (which is an eternity in browser development) a newer javascipt player might help, but that code a spaghetti mess.
I don't see a quick fix for this one, but does need some love.

s3m1s0n1c, 2021-01-03:
Mmm that's a bit shitty.. Oh well... it is what it is..

s3m1s0n1c, 2021-08-16:
I'll leave this one open..

Maybe someone might want the challenge to re-visit mplayer

bhall7, 2022-01-27:
This is still an issue. I noticed that audio from TR/Trunk-Player stops whenever the browser is minimized on iOS/iPad OS 15 (regardless of whether I'm using Safari, Chrome, or Firefox).

However, I did discover that the scanner app [NoraSector](https://app.norasector.com/) has figured out a way to keep the audio running in mobile web browsers even when the browser is not in focus. Not sure if it's because it's using WebRTC or something else.

MaxwellDPS, 2022-01-30:
@bhall7 this seems to be an issue with the Channels implementation of Websockets and websockets in general. I am working on a complete re-write to Trunk-Player and it will be using sockets.io... unfortunately for now there is not a great solution

closed 2024-03-20 by dreinhold