ScanOC/trunk-player #95: Audio files not playing

n71fs, 2021-02-26:
Managed to get my site setup today however I cannot listen to any of the audio files via the site or download them.

I've included a picture of my ssh which is showing the 404 errors being produced from this. I've gone through everywhere it asks for the audio file locations and verified they are set correctly.

The files that are being uploaded do have audio on them when I go through them. However I can't ever find the exact mp3s being referenced in the 404 errors.

Any ideas?

[image]

RadioTechGuy, 2021-02-26:
Are you using nginx as front end? 
ngnix should be serving these files, and the request should not be making it into django (as the logs) show.
Post you ngnix config

n71fs, 2021-02-26:
To be honest I'm not sure, I know just barely enough about what I'm doing to mess it all up lol, its been a lot of copy and pasting figuring this all out

And I believe this is what you are looking for.

```
# Sample nginx config file for trunk-player

# Enable upgrading of connection (and websocket proxying) depending on the
# presence of the upgrade field in the client request header
#map \$http_upgrade \$connection_upgrade {
#default upgrade;
#'' close;
#}

map $http_upgrade $connection_upgrade {
    default upgrade;
    ''      close;
}

# the upstream component nginx needs to connect to
upstream trunkplayer {
    # server unix:///path/to/your/mysite/mysite.sock; # for a file socket
    server 127.0.0.1:7055; # for a web port socket (we'll use this first)
}

server {
    listen         80 default_server;
    server_name    _
    charset     utf-8;

    client_max_body_size 75M;   # adjust to taste

    # audio_files
    location /audio_files  {
        alias /home/radio/trunk-player/audio_files/;
    }

    location /static {
        alias /home/radio/trunk-player/static/; # your Django project's static files - amend as required
    }

    location / {
        # Match PNG Icons
        location ~ /.*.png$ {
            root /home/radio/trunk-player/icons/;
        }
        # Pass request to the upstream alias
        proxy_pass http://trunkplayer;
        # Require http version 1.1 to allow for upgrade requests
        proxy_http_version 1.1;
        
        # We want proxy_buffering off for proxying to websockets.
        proxy_buffering off;
        
        # http://en.wikipedia.org/wiki/X-Forwarded-For
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        
        # enable this if you use HTTPS:
        # proxy_set_header X-Forwarded-Proto https;
        
        # pass the Host: header from the client for the sake of redirects
        proxy_set_header Host $http_host;
        
        # We've set the Host header, so we don't need Nginx to muddle
        # about with redirects
        proxy_redirect off;
        
        # Depending on the request value, set the Upgrade and
        # connection headers
        proxy_set_header Upgrade $http_upgrade;
        
        proxy_set_header Connection $connection_upgrade;
        
        proxy_set_header X-Forwarded-Proto https;
        
    }
}
```

RadioTechGuy, 2021-02-26:
Is trunk-recorder setup to put your audio files into /home/radio/trunk-player/audio_files/ ? Would be in the config.json

n71fs, 2021-02-26:
Correct, they are served there via this :

```
#! /bin/bash
# Script to upload audio from trunk-recorder to 
# web server to be displaied by trunk-player
#
# 02/15/2016 - Dylan Reinhold dreinhold@gmail.com
#-----------------------------------------------------
#
REMOTE_USER_NAME="radio"
REMOTE_SERVER="167.71.190.155"
REMOTE_AUDIO_FOLDER="/home/$REMOTE_USER_NAME/trunk-player/audio_files"
REMOTE_IMPORT_SCRIPT="/home/$REMOTE_USER_NAME/trunk-player/utility/trunk-player/encode_load.sh"
echo "Encoding: $1" 
filename="$1"
basename="${filename%.*}"
filename_only=$(basename $basename)
json="$basename.json"

# Hack the JSON to add play length
len=$(soxi -D $filename)

head -n-2 $json > $json.new
echo "\"play_length\": $len," >> $json.new
tail -n2 $json >> $json.new
mv $json.new $json

echo "Upload: $filename" 

scp -q -C ${filename} $json $REMOTE_USER_NAME@$REMOTE_SERVER:$REMOTE_AUDIO_FOLDER/
if [ $? -eq 0 ]
then
  ssh -q $REMOTE_USER_NAME@$REMOTE_SERVER "$REMOTE_IMPORT_SCRIPT $filename_only"
  echo "Remove files"
  rm -f $json $filename 
fi
```

RadioTechGuy, 2021-02-26:
Are you currently running it via the `./manage runserver` command?

n71fs, 2021-02-26:
Yes

n71fs, 2021-02-27:
Alright well I was messing around with the user sign up settings and now all the sudden its working. Don't think I edited anything remotely relevant to the serving of audio files, but I'll take it 🤣 .

Thanks for your help though.

n71fs, 2021-02-27:
Alright well right after I closed this I've begun encountering some issues again. I am still getting the following in the console 

```
Not Found: /home/radio/trunk-player/audio_files/17562-1614384443_859187500.mp3
2021-02-26 19:09:04,135 Not Found: /home/radio/trunk-player/audio_files/17562-1614384443_859187500.mp3
```
Seems like all/most are making it through initially though, Occasionally I cannot play or download one I just listened to a minute ago, despite having permissions to listen to much older calls.

dreinhold, 2021-02-28:
Well you made it this far, that is pretty good. There are a lot of moving parts that need to come together to make it all work. 
That is strange it some times works. So those files reported as missing, do they exist?
`ls /home/radio/trunk-player/audio_files/17562-1614384443_859187500.mp3` in the machine trunk-player is on.

Maybe the copy from trunk-recorder sometimes has issues?

I think we need to change document and move people to use the docker setup, but that's for another day ;)

n71fs, 2021-02-28:
I'll close this now, I ended up missing a setting for Audio_Files, somehow even after scouring those files for hours I missed it everytime. Now to figure out how to figure out why I can't add users 😆 I always get a duplicate key issue, and now just straight up 500 errors. 

Also as a sidenote personally I had issues getting docker setup, granted I have no clue what im doing so that might be part of it. I just ended up using your digital ocean YouTube tutorial as well as a couple other tutorials online.

closed 2021-02-28 by n71fs

bctrainers, 2021-02-28:
> Maybe the copy from trunk-recorder sometimes has issues?

Some setups have funky Linux filesystem limitations (too many files in a single folder resulting in phantom out-of-space errors), but this shouldn't be happening for a fresh install.

A long while back, I modified the generic uploader due to the aforementioned issue observed. It's a hack job by any standards, but it gets the job done. I added some sleep commands in this script due to processing lag observed on a virtual machine environment. Those may not be needed on a bare-metal/physical machine. This also needs some to-file logging support. :-)

I tried using a for/while loop on the JSON and wav file uploads originally, but ended up with some really funky issues (eg: It never got a proper status response, resulting in a never-ending loop)

```bash
#! /bin/bash
# Script to upload audio from trunk-recorder to
# web server to be displayed by trunk-player.
# 
# 2019: 
#     * Added basic sanity checks, with a sleeper between.
#     * Added json parse check for TG used for folder name.
#     * Split wav->mp3 conversion + upload and json upload process.
#
# 02/15/2016 - Dylan Reinhold dreinhold@gmail.com
# 05/01/2019 - Brett C brettc@replayirc.com
#-----------------------------------------------------
#

datestamp=`date '+%F %T.%N'`
fileloc=`date '+%Y/%m/%d/'`
REMOTE_USER_NAME="radio"
REMOTE_SERVER="192.168.20.212"
REMOTE_AUDIO_FOLDER="/home/$REMOTE_USER_NAME/trunk-player/audio_files"
REMOTE_IMPORT_SCRIPT="/home/$REMOTE_USER_NAME/trunk-player/utility/trunk-player/encode_load.sh"
filename="$1"
basename="${filename%.*}"
filename_only=$(basename $basename)
json="$basename.json"
sleep 1

# Check length, append to json file.
len=$(soxi -D $filename)
head -n-2 $json > $json.new
echo "\"play_length\": $len," >> $json.new
tail -n2 $json >> $json.new
mv $json.new $json
sleep 1

# Find the TG in use and append it to the web_url folder ID var
tgused=`jq -r '.talkgroup' $json`
fileloc=$tgused/`date '+%Y/%m/%d/'`

#  Open a SCP connection and upload the JSON first.
scp -qp $json $REMOTE_USER_NAME@$REMOTE_SERVER:$REMOTE_AUDIO_FOLDER/
scpstatus=$?
if [ $scpstatus -eq 0 ];
then
  sleep 1
  scp -qp $json $REMOTE_USER_NAME@$REMOTE_SERVER:$REMOTE_AUDIO_FOLDER/
else
  echo "General Failure Of JSON Upload Process - Failing $json ($scpstatus)"
fi

sleep 1

# Open another SCP connection, upload the wav file, let TrunkPlayer deal with file conversion.
scp -qp ${filename} $REMOTE_USER_NAME@$REMOTE_SERVER:$REMOTE_AUDIO_FOLDER/
scpstatus2=$?
if [ $scpstatus2 -eq 0 ];
then
  scp -qp ${filename} $REMOTE_USER_NAME@$REMOTE_SERVER:$REMOTE_AUDIO_FOLDER/
  sleep 1
  ssh $REMOTE_USER_NAME@$REMOTE_SERVER "$REMOTE_IMPORT_SCRIPT $filename_only $fileloc"
elif [ $scpstatus2 -ne 0 ];
then
  echo "[$datestamp][INFO] Upload process failed for $filename_only with error code $scpstatus2"
  echo "[$datestamp][INFO] Attempting the scp command again for $filename_only"
  scp -qp ${filename} $REMOTE_USER_NAME@$REMOTE_SERVER:$REMOTE_AUDIO_FOLDER/
  ssh $REMOTE_USER_NAME@$REMOTE_SERVER "$REMOTE_IMPORT_SCRIPT $filename_only"
else
  echo "General Failure Of WAV Upload Process - Failing $filename_only ($scpstatus2)"
fi
```