ScanOC/trunk-player #100: Restrict Menu Scan lists when not logged in

MaxwellDPS, 2021-05-31:
Howdy,

Would it be possible to restrict the scan lists menu and the talk groups menu from listing, when the user is not signed in?

s3m1s0n1c, 2021-08-16:
Sorry for the long delay..

You could use something like below in radio/templates/radio/site_live_nav.html
```
  {% if user.is_authenticated %}
 <ul id="menu-scanlist" role="menu" class="dropdown-menu">
               <li class = "divider"></li>
                 <li><a href="/userscan/">Custom Scan List</a></li>
            </ul>
         </li>
   {% endif %}
 ```  
  
  Off the top of my head.. I haven't tested it so u'll have to play around with it..
  
  Thanks
  Sonic

MaxwellDPS, 2021-09-25:
Duplicate of #63

closed 2021-09-25 by MaxwellDPS