ScanOC/trunk-player #108: Django3 branch TestCase are failing

dreinhold, 2021-09-15:
With the removal of pinax-stripe the test cases are not passing

Test cases are in radio/tests, this is another area that needs some love, need more test cases...

```

(venv) dylan@laptop:~/tp3$ ./manage.py test
Creating test database for alias 'default'...
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/core/management/__init__.py", line 419, in execute_from_command_line
    utility.execute()
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/core/management/__init__.py", line 413, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/core/management/commands/test.py", line 23, in run_from_argv
    super().run_from_argv(argv)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/core/management/base.py", line 354, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/core/management/base.py", line 398, in execute
    output = self.handle(*args, **options)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/core/management/commands/test.py", line 55, in handle
    failures = test_runner.run_tests(test_labels)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/test/runner.py", line 725, in run_tests
    old_config = self.setup_databases(aliases=databases)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/test/runner.py", line 645, in setup_databases
    debug_sql=self.debug_sql, parallel=self.parallel, **kwargs
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/test/utils.py", line 183, in setup_databases
    serialize=connection.settings_dict['TEST'].get('SERIALIZE', True),
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/db/backends/base/creation.py", line 79, in create_test_db
    run_syncdb=True,
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/core/management/__init__.py", line 181, in call_command
    return command.execute(*args, **defaults)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/core/management/base.py", line 398, in execute
    output = self.handle(*args, **options)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/core/management/base.py", line 89, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/core/management/commands/migrate.py", line 246, in handle
    fake_initial=fake_initial,
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/db/migrations/executor.py", line 117, in migrate
    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards
    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/db/migrations/executor.py", line 227, in apply_migration
    state = migration.apply(state, schema_editor)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/db/migrations/migration.py", line 116, in apply
    operation.state_forwards(self.app_label, project_state)
  File "/home/dylan/tp3/venv/lib/python3.7/site-packages/django/db/migrations/operations/fields.py", line 162, in state_forwards
    old_field = model_state.fields.pop(self.name)
KeyError: 'stripe_plan'

```

MaxwellDPS, 2021-09-16:
PR #109 should clear this up

dreinhold, 2021-09-17:
Fixed in 96fdc03

closed 2021-09-17 by dreinhold